Sloeber/arduino-eclipse-plugin #390: Fix the license

wimjongman, 2016-03-13:
I suggest to make the license official EPL [1].

[1] https://www.eclipse.org/legal/epl-v10.html

jantje, 2016-03-13:
I'm not sure we can do this.
There is some code that comes from other people. Don't they have to agree?
For instance https://github.com/jantje/arduino-eclipse-plugin/blob/master/it.baeyens.arduino.core/src/it/baeyens/arduino/tools/Boards.java#L28 comes originallly from trump
https://github.com/jantje/arduino-eclipse-plugin/tree/master/it.baeyens.arduino.core/src/cc/arduino/packages comes from arduino
and these are only 2 examples

wimjongman, 2016-03-13:
### Your License
The license you provide for your software is  your choice.

#### You created the file yourself
If you create it yourself, you choose the license. You also have to add a copy of the license in the header of the source.

#### You modified an existing source
If you modified another source, you cannot change the license. For example, if you changed the nebula oscilloscope which is EPL then you cannot move that source license to GPL

#### You include unmodified source (e.g. a library)
This is fine if their license says so. It does not say anything about how YOU license YOUR files.

### Things you need to do if you include other peoples software
You have to read the license and then do what that license requires. It could say a number of things:

* You maybe have to acknowledge them
* You maybe have to include their source
* Etc ..

For example this stuff from trump. You copied the source and changed it. It looks like this file is GPL but that should be re-examined. You have to make note of every file that you adapt from others and make clear that you acknowledge their license.

For example, if people want to include your stuff in a way that is okay with your license but not with trumps license then they should be able to see that in the license.txt file in the root of your repo. They could then choose to rewrite that part of the code or not us your code.

Also, you should always ensure that people that give you code must add the license header in their file and possibly also sign a CLA (example [1])

[1] http://contribute.jquery.org/CLA/

jantje, 2016-03-13:
I had typed a long answer here but I lost it.
Anyways the plugin has some past and Trump and the avr plugin are part of that past. Trump and I started this together and I have spoken with Thomas over the plugin.
I'm pretty sure that not each and every file contains the correct licence info (I'm pretty lazy at that) but I do not remove the original stuff and I try to reflect it in the folder structure. For Trump and Thomas that is not the case.
But you know better than I what this is all about and I'm perfectly ok with the epl licence.
So I say: "Gaan met die banaan"

wimjongman, 2016-03-14:
xxxx

wimjongman, 2016-04-02:
Fixed.

closed 2016-04-02 by wimjongman