Sloeber/arduino-eclipse-plugin #392: Custom Workspace variable for Arduino Plugin path for projects

mattb5906, 2016-03-20:
Would it be possible to create and assign a new Workspace variable such as ArduinoPluginPath with the path of the Arduino plugin? Then use this variable in the projects as a relative path for the includes such as core and variant? This would allow opening projects on different machines where the eclipse/plugin path is different. Or is there a better way?

Thanks

[image]
[image]

jantje, 2016-03-20:
The problem is a bit more complicated than this because there are multiple locations where the core can be located.
I actually proposed something in issue #34 that would solve this, but I never got any response.

mattb5906, 2016-03-21:
Thanks for the information Jantje. I had feeling it wouldn't be straight-forward. Out of interest, what other locations are there?

jantje, 2016-03-22:
You have private hardware and private library locations.
This is why V3 (in contrast to V2.X) no longer uses the path links

mattb5906, 2016-03-23:
Just another question: If I heavily modified the core libraries and wanted to move them to a private library and use those files instead of the automatically downloaded ones, how would I go about doing that? Would it be a case of manually editing the build environment values for the project? Thanks

mattb5906, 2016-03-23:
Forgot about the boards.txt file option. Changed to my local one and it that seems to work. One problem I encountered was that I had to create the A.RUNTIME.TOOLS.AVR-GCC.PATH variable.

jantje, 2016-03-25:
easiest is to build a jason file. Start from one that has the build and upload tools you need.
Then you can add your zip file and the  A.RUNTIME.TOOLS.AVR-GCC.PATH and A.RUNTIME.TOOLS.AVRDUDE.PATH should be defined
I'm closing this one as a duplicate of #34

closed 2016-03-25 by jantje