Sloeber/arduino-eclipse-plugin #393: [Error] Platform and Boards Manager (3.0.0-rc) fails to parse JSON board info

GHPS, 2016-03-22:
I was just trying to download the board info for my Digistump Digispark board
with the current development version of the plugin (commit c707e0e05117527dac4ad6d6ba843c3a390d768d)
from their official site (http://digistump.com/package_digistump_index.json).
The JSON file works fine in the Arduino IDE (1.6.5).

Unfortunately the parser fails to parse this file with the message:
```
it.baeyens.arduino.core
Error
Tue Mar 22 02:15:14 CET 2016
Unable to parse /usr/Arduino/arduino-eclipse-pluginV3/it.baeyens.arduino.product/target/products/it.baeyens.arduino.product/linux/gtk/x86/eclipseArduino/arduinoPlugin/package_digistump_index.json
```
The protocol gives the following details:
```
com.google.gson.JsonSyntaxException: java.lang.IllegalStateException: Expected BEGIN_OBJECT but was STRING at line 1 column 1
	at com.google.gson.internal.bind.ReflectiveTypeAdapterFactory$Adapter.read(ReflectiveTypeAdapterFactory.java:176)
	at com.google.gson.Gson.fromJson(Gson.java:803)
	at com.google.gson.Gson.fromJson(Gson.java:741)
	at it.baeyens.arduino.managers.Manager.loadPackageIndex(Manager.java:237)
	at it.baeyens.arduino.managers.Manager.internalLoadIndices(Manager.java:68)
	at it.baeyens.arduino.managers.Manager.loadIndices(Manager.java:176)
	at it.baeyens.arduino.ui.LinkPreferencePage.performOk(LinkPreferencePage.java:59)
	at org.eclipse.jface.preference.PreferencePage.performApply(PreferencePage.java:464)
	at org.eclipse.jface.preference.PreferencePage$2.widgetSelected(PreferencePage.java:294)
	at org.eclipse.swt.widgets.TypedListener.handleEvent(TypedListener.java:248)
	at org.eclipse.swt.widgets.EventTable.sendEvent(EventTable.java:84)
	at org.eclipse.swt.widgets.Display.sendEvent(Display.java:4481)
	at org.eclipse.swt.widgets.Widget.sendEvent(Widget.java:1329)
	at org.eclipse.swt.widgets.Display.runDeferredEvents(Display.java:3819)
	at org.eclipse.swt.widgets.Display.readAndDispatch(Display.java:3430)
	at org.eclipse.jface.window.Window.runEventLoop(Window.java:827)
	at org.eclipse.jface.window.Window.open(Window.java:803)
	at org.eclipse.ui.internal.dialogs.WorkbenchPreferenceDialog.open(WorkbenchPreferenceDialog.java:211)
	at org.eclipse.ui.internal.OpenPreferencesAction.run(OpenPreferencesAction.java:63)
	at org.eclipse.jface.action.Action.runWithEvent(Action.java:473)
	at org.eclipse.jface.action.ActionContributionItem.handleWidgetSelection(ActionContributionItem.java:595)
	at org.eclipse.jface.action.ActionContributionItem.access$2(ActionContributionItem.java:511)
	at org.eclipse.jface.action.ActionContributionItem$5.handleEvent(ActionContributionItem.java:420)
	at org.eclipse.swt.widgets.EventTable.sendEvent(EventTable.java:84)
	at org.eclipse.swt.widgets.Display.sendEvent(Display.java:4481)
	at org.eclipse.swt.widgets.Widget.sendEvent(Widget.java:1329)
	at org.eclipse.swt.widgets.Display.runDeferredEvents(Display.java:3819)
	at org.eclipse.swt.widgets.Display.readAndDispatch(Display.java:3430)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine$4.run(PartRenderingEngine.java:1127)
	at org.eclipse.core.databinding.observable.Realm.runWithDefault(Realm.java:337)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.run(PartRenderingEngine.java:1018)
	at org.eclipse.e4.ui.internal.workbench.E4Workbench.createAndRunUI(E4Workbench.java:156)
	at org.eclipse.ui.internal.Workbench$5.run(Workbench.java:694)
	at org.eclipse.core.databinding.observable.Realm.runWithDefault(Realm.java:337)
	at org.eclipse.ui.internal.Workbench.createAndRunWorkbench(Workbench.java:606)
	at org.eclipse.ui.PlatformUI.createAndRunWorkbench(PlatformUI.java:150)
	at org.eclipse.ui.internal.ide.application.IDEApplication.start(IDEApplication.java:139)
	at org.eclipse.equinox.internal.app.EclipseAppHandle.run(EclipseAppHandle.java:196)
	at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.runApplication(EclipseAppLauncher.java:134)
	at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.start(EclipseAppLauncher.java:104)
	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:380)
	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:235)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at org.eclipse.equinox.launcher.Main.invokeFramework(Main.java:669)
	at org.eclipse.equinox.launcher.Main.basicRun(Main.java:608)
	at org.eclipse.equinox.launcher.Main.run(Main.java:1515)
Caused by: java.lang.IllegalStateException: Expected BEGIN_OBJECT but was STRING at line 1 column 1
	at com.google.gson.stream.JsonReader.beginObject(JsonReader.java:374)
	at com.google.gson.internal.bind.ReflectiveTypeAdapterFactory$Adapter.read(ReflectiveTypeAdapterFactory.java:165)
	... 48 more
```
Session Data
```
eclipse.buildId=unknown
java.version=1.8.0_72-internal
java.vendor=Oracle Corporation
BootLoader constants: OS=linux, ARCH=x86, WS=gtk, NL=de_DE
Framework arguments:  -perspective it.baeyens.arduino.application.perspective
Command-line arguments:  -os linux -ws gtk -arch x86 -perspective it.baeyens.arduino.application.perspective
```

jantje, 2016-03-22:
I'll have a look.

jantje, 2016-03-22:
So I took a look. The problem you are seeing is because http://digistump.com/package_digistump_index.json is not a json file but a html file redirecting to https://raw.githubusercontent.com/digistump/arduino-boards-index/master/package_digistump_index.json
This concept is not supported by the plugin.
However you should get a nice and clean message. I'll look into that now.

jantje, 2016-03-22:
The last fix should have said #396

GHPS, 2016-03-23:
> This concept is not supported by the plugin.
> However you should get a nice and clean message. I'll look into that now.

Thanks for the fix.

Well - they send a "302 found but temporary moved" response which is a correct, standard compliant
way to do this.

The Arduino IDE supports this type of indirection so the plugin should do so as well.

Perhaps this piece of code can help implementing the feature:
http://stackoverflow.com/questions/20212505/handling-http-status-code-302-moved-temporarily-in-netty

jantje, 2016-03-23:
>The Arduino IDE supports this type of indirection so the plugin should do so as well.

Who says so?

jantje, 2016-03-25:
I added code to delete the bad jason files so you do not get in a sort of corrupted situation.
The link to stackoverflow you provided is not relevant in this case as the plugin uses a completely different download approach.

GHPS, 2016-03-28:
>Who says so?

Well the idea was that most people start with Digispark boards the same way as I did:
Read the Digispark Wiki (https://digistump.com/wiki/digispark/tutorials/connecting), copy the
json link into the Arduino IDE and download the board packages in the package manager.
Voila - works fine. 

But the same approach doesn't work in Eclipse.

jantje, 2016-03-28:
I decided to no longer respond to github issues -and other questions- due to a lack of support.
As a patron you can post a message on the patreon forum stating you support a github issue. This allows me to support people who support me.
https://www.patreon.com/posts/3700469

GHPS, 2016-11-23:
Just to be clear when installing the Digistump Digispark platform: Is this bug fixed or does the redirected link still corrupt the plugin?

jantje, 2016-11-23:
It is confusing I know.
Basically there are currently 2 nightly versions. This is due to the fact that V4 is install incompatible with V3.
So if you upgrade a V3 stable to the nightly you remain on V3.
As this issue is fixed in October this is fixed in V4 which is not yet released as stable. To get V4 you need to install from the nightly.
FYI V4 is stabilizing so I no longer dis-advice upgrading .

As an independent open source project we value your submission, but we ask for support, either by helping us out coding (yeah, we do understand it does require time) or a Patreon contribution (starting from as low as 1$ a month): this allows us to support people who support us back!
As a supporter you can drag our attention and prioritize your submission...
please become a supporter!
https://www.patreon.com/jantje

commit bd62ebfa (2016-10-12): After I fixed #393 I wanted to test it with the unit test program.

Out came a bug that if a recipe is left empty it causes problems.
So I fixed that.
It seemed that one of the boards changed its names in the menu options.
So I changed the option content from <menuName,MenuItemName> to
<menuID,MenuItemID>. 
Doing so I noticed the info was stored at 2 places (some key thingy and
in the environment variables). I went for only environment variables.
However environment variables are case insensetive -at least in the
implementation of Sloeber-
To fix this it was easier to start using BoardDescript [... cut]
closed 2017-01-17 by jantje