Sloeber/arduino-eclipse-plugin #394: Serial Monitor: Disable DTR

SuperOok, 2016-03-22:
When opening the serial monitor, the DTR and RTS pins are set to GND in order to reset the Arduino before flashing. That seems to be ok for Arduino, but not for an ESP8266. The ESP resets and goes directly into boot mode, so the serial monitor cannot be used.

In a visual Studio plugin for Arduino this is solved by adding an option to the serial monitor to disable the DTR grounding, see "DTR Checkbox" at
http://www.visualmicro.com/page/User-Guide.aspx?doc=Serial-Monitor.html

Can such option be added to the serial monitor of the eclipse IDE plugin?

jantje, 2016-03-22:
The serial monitor in the arduino eclipse plugin does not reset the arduino.
There is a reset button to reset the arduino if you want it (does not work with all boards).
I suspect you are experiencing another problem.

SuperOok, 2016-03-22:
Ok, thanks, I'll try to narrow it down.

SuperOok, 2016-03-22:
Ok, the problem is not the explicit reset, but the serial connection params themselves. You use 
```this.port.setParams(this.rate, this.databits, this.stopbits, this.parity);```
which implicitly seems to set RTS and DTR to true, so it's the same like
```serialPort.setParams(this.rate, this.databits, this.stopbits, this.parity, true, true)```
I tried different combinations and the serial connection only works fine for my board if I use
```serialPort.setParams(this.rate, this.databits, this.stopbits, this.parity, false, false)```

I tried with a standalone mini-test with the jssc library. Does it make sense to make the boolean flags configurable in the eclipse plugin?

jantje, 2016-03-22:
I made a comment on the comment of the pull request pffffffffffff :-)
I would not be surprised if arduino boards did not suffer from not pulling the dtr pin down as there is a capacitor. The arduino's do not reset when the plugin its serial monitor connects.
However it is a big risk to take just before release V3. As this fixes ESP8266 (which is really popular right now) I'd like this fix in V3. So I think the gui way is the best way right now. 
If you can quickly fix the comment I can accept the pull request and we have the fix tomorrow.

closed 2016-03-22 by jantje

SuperOok, 2016-03-23:
Thanks for fixing the comment, (was just about to do it) and thanks for merging!

jantje, 2016-03-23:
Is your problem fixed now with the nightly build?

XeptionE, 2020-12-06:
Hi, I still have this problem with my 

Eclipse IDE for C/C++ Developers (includes Incubating components)
Version: 2020-09 (4.17.0)
Build id: 20200910-120

and

C/C++ Development Tools
Version: 9.11.1.202006011430
Build id: 20200601-1605

and

C/C++ Arduino Tools
Version: 9.11.1.202005042001
Build id: 20200601-1605

In the "Serial monitor view" if I click on the + Button, a dialog pops up with serial port and baudrate combobox. Also a checkbox is there "Set DTR and RST". It doesn't matter, if this checkbox is enabled or disabled. The ESP resets and remains reseted until I close the serial connection.

I was very happy using the previously in eclipse integrated Arduino support. Everything worked fine, serial monitor was fine (even it was deactivated while uploading software to ESP and reactivated after upload etc.). But I missed the possibility to upload via OTA. Now OTA is possible, but many other things are broken, especially this problem here with the console.

jantje, 2020-12-06:
@XeptionE 
Sloeber and "Eclipse IDE for C/C++ Developers (includes Incubating components)' are different tools
So you are at the wrong adress

XeptionE, 2020-12-07:
Oh sorry, I though slober were the plugin in background, because there is a menu point "Arduino/Show the Known limitations of Slober..." :)

jantje, 2020-12-07:
Oeps I misread "Eclipse IDE for C/C++ Developers" as "Eclipse IDE for Arduino". Sorry for that
I don't know what "C/C++ Arduino Tools" is.
You also do not mention the sloeber version you are on. If you are on the nightly please report your issue in #1261 as that is most likely the root cause.
This is the first change in a long time in the serial area.

XeptionE, 2020-12-07:
My slober plugin was installed from marketplace and has the version 4.3.3.202005042010.

But in marketplace I see two installed tools:
"The Arduino Eclipse IDE named Sloeber (Product) V4"
"The Arduino Eclipse plugin named Sloeber V4"

Which is the right one? Or is the one the plugin and the other only Eclipse-Product? The last time I was active developer for Eclipse plugins is ~15 years ago :)

XeptionE, 2020-12-07:
Another hint for my serial problem:

If I open the serial port using Putty, the same behaviour occurs: ESP resets and "hangs" on reset until I close the putty connection. This is not the case using Arduino UI nor using an older eclipse CDT version (with support for Arduino) and their Terminal window.

On my  newest eclipse CDT the terminal window produces an exception and it stays empty view so I wanted to use the Serial monitor view instead. But this resets ESP and (if I disconnect the enable pin from serial) it steals the COM port not allowing to upload images. I always need to disconnect manually first and then upload and then reconnect.

tx0h, 2021-02-03:
hi,

when i use minicom (linux) and exit minicom without reset (ctrl-a q), everything is fine as well.

for the eclipse arduino ide it would be nice if i could set DSR/DTS and RTS/CTS somewhere in the preferences instead of select/deselect it at each new connection.

it would be helpful to either select:
* DSR/DTS
* RTS/CTS
* DSR/DTS + RTS/CTS
* none

in the https://esp32.com/viewtopic.php?t=4988 forum, there is a post, who switched from DSR/DTR to RTS/CTS to solve the problem with the resets.

i hope, this helps.

tx0h

jantje, 2021-02-03:
@tx0h 
I think I like your proposal.
However this is an old issue closed 4 years ago. Please create a new issue.

PR #395 (2016-03-22): Fix for #394, adding config parameter for RST/DTR flags in serial connection
Added a boolean flag for a serial connection (COM Port) to configure the RST/DTR flags of the serial connection.
I testet it and can confirm it fixes #394 for me.