Sloeber/arduino-eclipse-plugin #397: V3 core dump/immediate crash by clicking on "Where do I find json URL's?"

GHPS, 2016-03-22:
I did some further testing on the Platform and Boards manager. 

No good idea was using the offered help: In V3 I get an immediate crash and core dump of Eclipse by clicking on "Where do I find json URL's?".

```
#
# A fatal error has been detected by the Java Runtime Environment:
#
#  SIGSEGV (0xb) at pc=0x60c85005, pid=10226, tid=3075372800
#
# JRE version: OpenJDK Runtime Environment (8.0_72-b15) (build 1.8.0_72-internal-b15)
# Java VM: OpenJDK Server VM (25.72-b15 mixed mode linux-x86 )
# Problematic frame:
# C  [libgdk-x11-2.0.so.0+0x48005]  gdk_display_open+0x45
#
# Failed to write core dump. Core dumps have been disabled. To enable core dumping, try "ulimit -c unlimited" before starting Java again
#
# An error report file with more information is saved as:
# /usr/lib/arduino-eclipse-pluginV3/it.baeyens.arduino.product/target/products/it.baeyens.arduino.product/linux/gtk/x86/eclipseArduino/hs_err_pid10226.log
#
# If you would like to submit a bug report, please visit:
#   http://bugreport.java.com/bugreport/crash.jsp
# The crash happened outside the Java Virtual Machine in native code.
# See problematic frame for where to report the bug.
#
```

jantje, 2016-03-22:
It should open a browser :-(
Do you get the same issue when you do menu->arduino->learn/"how to"/video's ?

GHPS, 2016-03-22:
Yes - all external links cause a complete crash.

BTW: The L in "learn" is lower case. And "Video's" should be "Videos" (simple plural s).

GHPS, 2016-03-22:
> It should open a browser :-(

There's a setting under "General/Web Browser". 

By default it is set to "internal". I set it to "external" and "Firefox" instead of "Default system web browser".

Didn't help either - still crashing.

GHPS, 2016-03-23:
Update: The problem is definitely related to the Arduino plugin: If I open Help/Eclipse Marketplace/Newsletter I can click on any link to read it in my web browser - without crash.

jantje, 2016-03-23:
It works fine under windows 10 and linux redhat.
If I'm not mistaken you use ubuntu 14. I'll have a look there.
Eclipse settings will not impact this as I use a standard java way of working.

```
String url = event.getParameter("it.baeyens.arduino.core.link.parameter");//$NON-NLS-1$
Desktop.getDesktop().browse(new URI(url));
```

https://github.com/jantje/arduino-eclipse-plugin/blob/master/it.baeyens.arduino.core/src/it/baeyens/arduino/actions/LinkCommandHandler.java#L24

jantje, 2016-03-25:
I just tried this in ubuntu 14 and eclipse freezes and then crashes with a very similar messages as you provided.
I'm looking for the root cause now.

jantje, 2016-03-25:
I'm now using org.eclipse.swt.program.Program.launch
I hope this fixes it. Form tomorrow onwards available on the nightly for tests

jantje, 2016-03-26:
It worked for the menu but not for the json link.
Can someone try the new code on Mac?

jantje, 2016-03-27:
I tested this on ubuntu 14 and on windows 10. 
It works now.
It would be nice if someone can confirm it also works on mac

universpharmacie, 2016-03-27:
Just updated to the latest nightly under Mac OSX 10.11.4 and tried to click onto Arduino => How To..., Learn..., and Videos...
Each time, Safari opens and the correct content is displayed. So I can confirm that it is fixed for Mac, too.

jantje, 2016-03-27:
@universpharmacie 
txs :-)

universpharmacie, 2016-03-27:
you're (as always) welcome!

GHPS, 2016-03-28:
I can confirm that the bug is fixed under Kubuntu 14.04 LTS.

Jantje - you nailed it. Great, perfect.

jantje, 2016-03-28:
I keep issues open till they are in a release.
The flag fixed in nightly is used to show the nightly status.

closed 2016-04-01 by jantje