Sloeber/arduino-eclipse-plugin #398: V3 empty checkbox even is subitem is selected

GHPS, 2016-03-23:
Problem: Just a minor annoyance: In Platforms and Boards the checkboxes currently don't present the correct state of the selection. An empty checkbox (meaning "No selection") is shown even if a subitem is selected.

Solution: The checkmark with a dotted line checksymbol - to indicate that at least one subitem was chosen.

Further question: Does the plugin allow the selection of multiple versions of a board description at the same time, e.g. selecting version 1.6.10 and version 1.6.9?

If not a dropdown box or a radio button would be more intuitive.

jantje, 2016-03-23:
>Further question: Does the plugin allow the selection of multiple versions of a board description at the same time, e.g. selecting version 1.6.10 and version 1.6.9?

Yes
For libraries?
No

That is why the libraries have a dropdown for version selection and the boards have checkboxes.

We know that the look and feel could be better but we wanted to release the functionality and then work on making it nicer.

jantje, 2016-10-07:
This is also fixed as part of the fix #532

closed 2017-01-17 by jantje