Sloeber/arduino-eclipse-plugin #399: V3 Board.txt file is referenced by fully qualified path and file name

GHPS, 2016-03-23:
Problem: In the Project/Properties/Arduino dialog a different board.txt file can be selected. Unfortunately the dialog gives you a fully qualified path and file name - with the relevant information at the end. You have to make the dialog wide enough to see the last piece of information. In my (rather bad) case the dialog has to be 1862 pixels wide to be usable.

[image]

Solution: Shorten the dialog to the information already available in the Platform and Boards Manager: 
arduino/Arduino AVR/Boards/1.6.10
arduino/Arduino SAM Boards/1.6.7

jantje, 2016-03-23:
I noticed this myself. As you figured out yoursel the work around is to use a smaller path for the arduinoplugin storage.

>Solution: Shorten the dialog to the information already available in the Platform and Boards Manager:
arduino/Arduino AVR/Boards/1.6.10
arduino/Arduino SAM Boards/1.6.7

That won't work as it doesn't really work that way.

jantje, 2016-03-25:
In case someone wonders why this is not applicable for windows: In windows the path can not be longer then something around 256. Therefore the plugin issues a warning when running from a long path reducing the possible impact of this issue.

GHPS, 2016-03-28:
> That won't work as it doesn't really work that way.

OK - Can you please elaborate on these points?

jantje, 2016-03-28:
I decided to no longer respond to github issues -and other questions- due to a lack of support.
As a patron you can post a message on the patreon forum stating you support a github issue. This allows me to support people who support me.
https://www.patreon.com/posts/3700469

universpharmacie, 2016-03-28:
With a sardonic smile: Who cares about Windows? That's not an OS for true developers... :D

jantje, 2017-01-05:
@mjmeijer thanks for fixing this issue.

closed 2017-01-17 by jantje