Sloeber/arduino-eclipse-plugin #401: ESP8266 version issue

mdilger, 2016-03-29:
Hi Guys,

I am using the latest nightly build (win64.2016-03-28_04-21-38.tar.gz) and have an issue loading  a simple blink sketch to the ESP8266.

When I use the esp8266 1.6.5 platform it works ok, i.e.compiles and uploads to the ESP8266.

However when I use v2.1.0 I get the following message.

20:26:18 **** Incremental Build of configuration Release for project Blink ****
"C:\\Eclipse\\eclipseArduino\\arduinoPlugin\\tools\\make\\make" all 
'Building file: ../Blink.cpp'
'Starting C++ compile'
"C:\Eclipse\eclipseArduino\arduinoPlugin\tools\esp8266\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:/Eclipse/eclipseArduino/arduinoPlugin/packages/esp8266/hardware/esp8266/2.1.0/tools/sdk/include" -c -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11 -MMD -ffunction-sections -fdata-sections -DF_CPU=80000000L  -DDEBUG_ESP_CORE -DDEBUG_ESP_SSL -DDEBUG_ESP_WIFI -DDEBUG_ESP_HTTP_CLIENT -DDEBUG_ESP_HTTP_UPDATE -DDEBUG_ESP_HTTP_SERVER -DDEBUG_ESP_UPDATER -DDEBUG_ESP_OTA -DDEBUG_TLS_MEM -DARDUINO=10606 -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266  -DESP8266   -I"C:\Eclipse\eclipseArduino\arduinoPlugin\packages\esp8266\hardware\esp8266\2.1.0\cores\esp8266" -I"C:\Eclipse\eclipseArduino\arduinoPlugin\packages\esp8266\hardware\esp8266\2.1.0\variants\generic" -MMD -MP -MF"Blink.cpp.d" -MT"Blink.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "../Blink.cpp"  -o  "Blink.cpp.o"   -Wall
subdir.mk:18: recipe for target 'Blink.cpp.o' failed
process_begin: CreateProcess(NULL, C:\Eclipse\eclipseArduino\arduinoPlugin\tools\esp8266\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++ -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ -IC:/Eclipse/eclipseArduino/arduinoPlugin/packages/esp8266/hardware/esp8266/2.1.0/tools/sdk/include -c -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11 -MMD -ffunction-sections -fdata-sections -DF_CPU=80000000L -DDEBUG_ESP_CORE -DDEBUG_ESP_SSL -DDEBUG_ESP_WIFI -DDEBUG_ESP_HTTP_CLIENT -DDEBUG_ESP_HTTP_UPDATE -DDEBUG_ESP_HTTP_SERVER -DDEBUG_ESP_UPDATER -DDEBUG_ESP_OTA -DDEBUG_TLS_MEM -DARDUINO=10606 -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266 -DESP8266 -IC:\Eclipse\eclipseArduino\arduinoPlugin\packages\esp8266\hardware\esp8266\2.1.0\cores\esp8266 -IC:\Eclipse\eclipseArduino\arduinoPlugin\packages\esp8266\hardware\esp8266\2.1.0\variants\generic -MMD -MP -MFBlink.cpp.d -MTBlink.cpp.o -D__IN_ECLIPSE__=1 -x c++ ../Blink.cpp -o Blink.cpp.o -Wall, ...) failed.
make (e=2): The system cannot find the file specified.

make: *** [Blink.cpp.o] Error 2

20:26:18 Build Finished (took 423ms)

Does anyone have any idea what is wrong or am I doing something wrong, which is quite possible.

The reason I want to use v2.1.0 is for a particular application where the MQTT library specifies this.

Best regards,
Mike Dilger
p.s. Yes, I am a patron ;-)

jantje, 2016-03-29:
I'll give it a shot here and I'll report back

jantje, 2016-03-29:
It works here
Are you sure you waited long enough for the boards and toolchains to download.
Can you specify the exact board that fails?

mdilger, 2016-03-29:
Thanks for checking, I will take another look tonight and come back with more detail if I can't resolve the issue.
Cheers,
Mike

cyberreefguru, 2016-03-29:
This happens to me occasionally (I'm still on 2.4).  If you open preference, select board manager, and let the plugin reconfigure itself the issue generally goes away.  I don't know what causes it - I have not had time to look into it and I cannot reproduce it reliably (*seems* to happen after exiting and restarting).

mdilger, 2016-03-29:
Thanks for the feedback.  I will check this further tonight.
Cheers,
Mike

jantje, 2016-03-29:
@cyberreefguru 
There is no boards manager in 2.4 so I guess you are on a nightly.
What do you mean with "let the plugin reconfigure itself"?

cyberreefguru, 2016-03-29:
I'm on Luna with Arduino 1.6.5;   Arduino Eclipse Extensions 2.4.0.201508050214

I might be on nightly; I don't recall that, but it's been a long time since I futzed with the plugin.  Since it's working fine for my purposes, I'm not screwing with it.

Project -> Properties -> Arduino -> Boards.txt -> NodeMCU 1.0 -> Apply (Arduino app launches, configs, closes) -> OK

mdilger, 2016-03-29:
Jantje,

i have been thinking about your comment above and maybe the toolchain for esp8266 v2.1.0 has not downloaded correctly.

On eclipseArduino v3.0.0 what is the best way to remove the current installed esp8266 board platforms and then start again.  Do I simply delete the esp8266 folder structure from 'tools' or is there more that I need to do to remove it?  After that go to preferences and reselect the esp8266 options.

Cheers,
Mike

mdilger, 2016-03-30:
OK, I have sorted the compile problem by deleting and then reloading the board platforms.  I am now dealing with a slightly different issue with regard to uploading to the ESP8266.  I will close this issue and then work on the next one.  Thanks you the helpful suggestions.  Cheers, Mike

closed 2016-03-30 by mdilger

jantje, 2016-03-30:
@cyberreefguru 
Based on your second remark I'd say you are on arduino eclipse plugin 2.4. Which means you are on the latest stable which is the best if you want to get work done and it is good enough for you.
I was confused as you mentioned the boardsmanager as if it was part of the plugin. 
FYI what you called "let the plugin reconfigure itself"
We call that "Reconfigure the project"

jantje, 2016-03-30:
@mdilger 
To remove the installed boards simple deselect it in the windows->preferences->arduino->boards page and select OK.
You can reinstall it directly after that by selecting the board again in windows->preferences->arduino->boards page and select OK.

**Important note:** The installation process consists of 2 steps: download and unzip.
Due to performance and bandwidth reasons the process described above will only undo the unzip and redo the unzip. In other words tha download file is and remains on your system.
In case the downloaded file is corrupted you need to delete the dowloaded file manually. The downloaded files are located in the [eclipse install location]/arduinoPlugin/downloads folder

cyberreefguru, 2016-03-31:
@mdilger  FYI - I could not get esp8266 v2.1.0 to work properly with my version of the plugin. I had to stick with 2.0.

mdilger, 2016-03-31:
@cyberreefguru Thanks for the feedback.  I will try some more experimentation tonight.  I think 2.1.0 is now compiling Ok, but the upload to the esp8266 is failing.  I will try 2.0 as well and confirm.

cyberreefguru, 2016-03-31:
@mdilger I could not get 2.1 to compile - all sorts of path problems.  The plugin was pointing to the 2.0 directory, which I never downloaded (I was only 1.6 I think and jumped to 2.1).  Once I got 2.0 working, I left it alone.  0.02.  YMMV.

jantje, 2016-03-31:
@cyberreefguru FYI
Your story matches my expectations.
The root cause of your story are the changes done in the arduino ide when the arduino core team added the boardsmanager (I'm not saying it is a bad decision). The boardsmanager has a huge impact on what is described in the platform.txt (read platform.txt and boards.txt) and more importantly what is not.
Before the boardsmanager it was like: it is described in platform.txt or it is "hardcoded"
With the boardsmanager it is: It is described in the json or the platform.txt or it is hardcoded.
These issues hit the non AVR Arduino boards the most. 
2.4 queries the Arduino IDE for the json part. It stores the query results in DUMP files in the root of your workspace. Keeping these DUMP files up-to-date is really difficult.
I think we did a good job to fix these issues with V3 (release tomorrow) by implementing our own boardsmanager.

cyberreefguru, 2016-03-31:
@jantje thanks.  I've often wondered what those dump files are, and frankly, I've been deleting them since I like a tidy root directory.  Perhaps that is what triggers the need to re-configure the settings.

jantje, 2016-04-01:
@cyberreefguru 
Indeed it does.
And V3.0 no longer has them.
And V3.0 is released :-)
Time to upgrade :-))))

mdilger, 2016-04-01:
@cyberreefguru  I managed to solve my uploading issue by using the earlier version of esptool.exe (0.4.5) with v2.1.0.  This is of course running on v3.0.0 of eclipseArduino. Cheers, Mike