Sloeber/arduino-eclipse-plugin #402: Latest nightly build issues 3.0.0.201603280209

mattb5906, 2016-03-29:
Just for info: Updated to the latest nightly build, but there were issues during the startup.
[image]
[image]

jantje, 2016-03-29:
The first warning is because we detected that when you have sh or bash in your path there is a problem with the plugin. Nice to see it works ;-)
I'm not sure what the other error is about. But it does not look arduino eclipse plugin related to me.

mattb5906, 2016-03-30:
Just after I posted that I saw your latest commit comment mentioning the sh warning addition.

I had troubles reverting, so in the end I deleted the whole plugin and reinstalled to see if it fixed the Eclipse internal errors. Unfortunately, both errors remained so I deleted sh.exe from the Win AVR directory that my PATH variable pointed to. AFter doing this both types of errors went away on the next launch. As soon as sh is reinstated both types of errors come back.

jantje, 2016-03-30:
Txs for the tests and feedback.
After we found a major issue with sh and or bash in the path I added this test. Seems like there are more issues related to these files in the path on windows.

closed 2016-04-01 by jantje

mattb5906, 2016-04-04:
Not sure if this is intentional, but I've updated to the latest build and restored sh.exe and found that Eclipse now fails to launch after clicking OK on the first error message.

jantje, 2016-04-04:
Nothing changed in this area. So if it is the same sh.exe it should behave the same.
I do call "sh -version" where it might be better to call "where sh"
I assume the first error message is the one telling you sh is in the path?
What do you mean by fail?

mattb5906, 2016-04-04:
Thanks Jantje.
Yes, it was to do with sh warning. I've since had some big problems with sending uploading the hex files via avrdude. It did the touch process, found the new com port and tried to launch avrdude but it went straight to fail as though it didn't even attempt an upload. I tried a clean project with the default ino file which also failed, so I then tried it using the Arduino IDE where it worked first time and consistently.

To rule everything out I uninstalled the plugin from Eclipse, closed Eclipse and deleted the arduinoPlugin folder. I then restarted Eclipse and reinstalled from the nightly build site. All of this was while sh.exe was missing from the directory. After the install had completed and I tried to build a new project, it failed with the following:

```
16:19:44 **** Incremental Build of configuration Release for project NewTest ****
"C:\\eclipse\\arduinoPlugin\\tools\\make\\make" all 
'Building file: ../.ino.cpp'
'Starting C++ compile'
"/bin/avr-g++" -c -g -Os -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10606 -DARDUINO_AVR_MICRO -DARDUINO_ARCH_AVR  -DUSB_VID=0x2341 -DUSB_PID=0x8037 -DUSBCON  -DUSB_MANUFACTURER=\""Unknown"\" -DUSB_PRODUCT=\""Arduino Micro"\"   -I"C:\Matt\Arduino\Hardware\avr\1.6.9\cores\arduino" -I"C:\Matt\Arduino\Hardware\avr\1.6.9\variants\micro" -MMD -MP -MF".ino.cpp.d" -MT".ino.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "../.ino.cpp"  -o  ".ino.cpp.o"   -Wall
subdir.mk:24: recipe for target '.ino.cpp.o' failed
process_begin: CreateProcess(NULL, sh.exe -c "/bin/avr-g++ -c -g -Os -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10606 -DARDUINO_AVR_MICRO -DARDUINO_ARCH_AVR -DUSB_VID=0x2341 -DUSB_PID=0x8037 -DUSBCON -DUSB_MANUFACTURER=\"Unknown\" -DUSB_PRODUCT=\"Arduino Micro\" -IC:\Matt\Arduino\Hardware\avr\1.6.9\cores\arduino -IC:\Matt\Arduino\Hardware\avr\1.6.9\variants\micro -MMD -MP -MF.ino.cpp.d -MT.ino.cpp.o -D__IN_ECLIPSE__=1 -x c++ ../.ino.cpp -o .ino.cpp.o -Wall", ...) failed.
make (e=2): The system cannot find the file specified.

make: *** [.ino.cpp.o] Error 2

16:19:45 Build Finished (took 605ms)
```

When I added sh.exe back to the directory the error message was displayed again (as expected) but this time Eclipse opened correctly. I think part of the problem may have been down to the the update procedure rather than doing a fresh install.

With sh.exe added the build process still fails.

Maybe it's something obvious that I'm missing.

mattb5906, 2016-04-04:
Actually I was mistaken, after relaunching again with sh restored it brought back the "multiple problems have occurred" error box.

mattb5906, 2016-04-04:
Right, so I decided to eliminate my Eclipse install (Mars2) and downloaded a fresh version of Neon M6. I then installed the a fresh copy of the nightly plugin build. No more earlier mentioned build problems :+1: 

And after much testing, I've found out that my upload problems must be to do with the latency of a USB hub as AVRDUDE just hangs there during an upload attempt when connected to the hub.

When sh is back, it still causes issues with Eclipse sometimes it launches and gives errors and other times it just fails to launch all-together/crashes. I think this is probably how my version of Mars2 became buggy,

I will just work around this and keep sh removed (not used WInAVR in a long time anyway).
Cheers

rlogiacco, 2016-04-09:
Just a note: you don't have to delete the `sh.exe` file, you just have to remove it from the system PATH. Another problem maker can be Git client in case you decide to add Git binaries to your PATH: there is an `sh.exe` in there too.

jantje, 2016-04-09:
There is a follow up fix for this #415