Sloeber/arduino-eclipse-plugin #408: V3 "Automatically import libraries based on includes" setting does not get saved

compihu, 2016-04-02:
## How to reproduce:
- Start V3.0.0
- create a new workspace
- crate any arduino project (not really necessary)
- go to (menu)Window/Preferences->Arduino
- untick "Automatically import libraries based on includes"
- click Apply, then OK (or only OK)
- exit the IDE
- start the IDE again with the same workspace
- check the setting you unticked before exit -> option is active again

Tested on Linux && Windows

jantje, 2016-04-02:
txs I'll have a look.
Any reason why you want to deactivate it?

compihu, 2016-04-02:
Thanks,
The reason is that I use custom/modified/bugfixed versions of libraries included in the project tree as git subtrees.
You can take a look at it @ https://github.com/compihu/gate433 - fairly complex project layout

jantje, 2016-04-02:
You should use the import source code for that. Use another folder then libraries as that folder is maintained by the plugin.

compihu, 2016-04-03:
I have a shared virtual folder for the libraries named **"libs"** so there is no name collision either with the old **"Libraries"** nor the new **"libraries"**. The problem is that the names of the include files are the same as in the "stock" libraries, so the plugin "helps" me adding the default libraries to my project which I have to avoid.

Btw adding "**Automatically\ import\ libraries=false**" line to  **{WORKSPACE}/.metadata/.plugins/org.eclipse.core.runtime/.settings/it.baeyens.arduino.prefs** works it around until some other action removes or changes it to true (happened few times while tried to find a workaround)

jantje, 2016-04-03:
txs for the workaround.

> I have a shared virtual folder for the libraries named "libs" so there is no name collision either with the old "Libraries" nor the new "libraries". The problem is that the names of the include [...]

I think we are talking about different things.

jantje, 2016-04-03:
I confirm this is not working as designed.

closed 2016-08-01 by jantje

jantje, 2017-08-21:
This is a really old issue.
Hardware libs overwriting private ones is by design.

Mick59, 2017-08-21:
Thanks for the answer.

The problem is that in this way Eclipse is too locked for my needs. Not
easy to tweak some Arduino libs without renaming them.
It would be really useful if the library and core directories in the
project directory got the priority.

Regards,
Michael

jantje, 2017-08-21:
If you want to overrule you can always use the "add source code to the project".
Once a header is found Sloeber will no longer add the lib.