Sloeber/arduino-eclipse-plugin #409: V3 "Private Library path" not available for "Add[ing] a library to the selected project" when more than one path in the preferences

nafep, 2016-04-03:
When having **more than one** path as "_Private library path_" in the plugin's general preferences, none of the libs in those folders is available for selection in the "_Add a library to the selected project_" list after having quit and restarted Eclipse. (Everything works fine with only one path.)

[Workaround] To make the private libraries available again, it suffices (once per Eclipse session) to go to Preferences>Arduino and press "OK". (Note that the names of the private library folders are still there; there's no need to type them in again.)

Tested on a Mac.

jantje, 2016-04-03:
The multi private library functionality is one of my favourite additions (using multiple version control systems at the same time) so I tested this one right away and it works fine on windows.
Do you see any errors in the error log?

nafep, 2016-04-03:
The error log is desperately empty.

jantje, 2016-04-03:
Wierd
As I don't have a mac, it is working on windows and we have no logging; I need someone else to debug on a mac to find the root cause.

nafep, 2016-04-03:
Okay, understood. 
I tried to take a look at the code by myself, compiling it according to your description, but I'm unfortunately way too incompetent to solve the issue.
Although I might continue giving it a shot once in a while, I guess we'll be at least two waiting for someone who can reproduce and solve the issue. I'll be using the workaround in the mean time. 
Thanks for your support and the great plugin!

EdwinCroissant, 2016-04-08:
I am on the nightly build with Windows 10 and had the same problem until today.
It looks like the last change fixed it for me. Thank you for your good work :)

closed 2016-08-01 by jantje