Sloeber/arduino-eclipse-plugin #410: 32 bits eclipseArduino does not start under Ubuntu 16.04 64 bits with openjdk 1.8

Pilleo, 2016-04-03:
here is the error:
`JVM terminated. Exit code=13
/usr/bin/java
-Dosgi.requiredJavaVersion=1.8
-Dorg.eclipse.update.reconcile=false
-Xms40m
-Xmx512m
-jar /home/leonid/Запампоўкі/eclipseArduino (2)//plugins/org.eclipse.equinox.launcher_1.3.100.v20150511-1540.jar
-os linux
-ws gtk
-arch x86
-showsplash
-launcher /home/leonid/Запампоўкі/eclipseArduino (2)/eclipseArduinoIDE
-name EclipseArduinoIDE
--launcher.library /home/leonid/Запампоўкі/eclipseArduino (2)//plugins/org.eclipse.equinox.launcher.gtk.linux.x86_1.1.300.v20150602-1417/eclipse_1612.so
-startup /home/leonid/Запампоўкі/eclipseArduino (2)//plugins/org.eclipse.equinox.launcher_1.3.100.v20150511-1540.jar
--launcher.appendVmargs
-exitdata d00015
-perspective it.baeyens.arduino.application.perspective
-vm /usr/bin/java
-vmargs
-Dosgi.requiredJavaVersion=1.8
-Dorg.eclipse.update.reconcile=false
-Xms40m
-Xmx512m
-jar /home/leonid/Запампоўкі/eclipseArduino (2)//plugins/org.eclipse.equinox.launcher_1.3.100.v20150511-1540.jar `

jantje, 2016-04-03:
As far as I know you can not run 32 bit applications in a 64 bit linux.

Pilleo, 2016-04-03:
You perfectly can. There is no 32 bit skype under linux, but i am using it all the time.

jantje, 2016-04-03:
Have you installed java 8 32 bit as well?

Pilleo, 2016-04-03:
I am using standart ubuntu repository, and i cannot see 32 bit java there. Just openjdk/jvm, that is all.

jantje, 2016-04-03:
I decided to no longer respond to github issues -and other questions- due to a lack of support.
As a patron you can post a message on the patreon forum stating you support a github issue. This allows me to support people who support me.
https://www.patreon.com/posts/3700469

Pilleo, 2016-04-03:
It is ok, i can understand it. But if it is common for Ubuntu, than you will have a very little support from Ubuntu users.

compihu, 2016-04-04:
sudo apt-get install openjdk-8-jre:i386 maybe?

Pilleo, 2016-04-04:
Tried `sudo apt-get install openjdk-8-jre:i386`
Same result.

compihu, 2016-04-04:
Then simply use the version matches your 64 bit OS.

Pilleo, 2016-04-04:
Well, it works to be honest. But 64 bit version has another bugs why i cannot use it)

jantje, 2016-04-04:
Ubuntu 16.04 is still in beta.

closed 2016-04-04 by jantje