Sloeber/arduino-eclipse-plugin #411: 64 bit eclipseArduino on Ubuntu 16.04 64bit with openjdk 1.8 high cpu load

Pilleo, 2016-04-03:
CPU load is very high. If i try to manage libraries, than it freezes, and i have to close eclipse. It starts  ok after that, but library manager is hardly usable.

jantje, 2016-04-03:
The manage libraries takes a while when opening. During that eclipse is frozen. We are working on that.

Pilleo, 2016-04-03:
Unfortunately cpu load is high all the time, but libraries make it fatal. Also i was not able to load sketch to arduino uno, but it could build it. I was making everything like you did in your tutorial video. Also avrdude shows 0% all the time after i tried to upload sketch to arduino. Also i could not find anything about autocompletion in arduinoEclipse, does(should) it work? On my laptop it does not.

jantje, 2016-04-03:
I decided to no longer respond to github issues -and other questions- due to a lack of support.
As a patron you can post a message on the patreon forum stating you support a github issue. This allows me to support people who support me.
https://www.patreon.com/posts/3700469

jantje, 2016-04-04:
Ubunto 16.0 is still in beta.

closed 2016-04-04 by jantje