Sloeber/arduino-eclipse-plugin #412: First start failing

dhrupal93, 2016-04-04:
When i first started, as shown below. V3
[image]
it shows following error:
[image]

And then shows:
[image]

Note: Windows 10
EclipseArduino V3 (Downloaded on 3rd April 2016)

jantje, 2016-04-04:
It looks like something went wrong when downloading the file http://downloads.arduino.cc/packages/package_index.json
Can you dump the content of [eclipse installation]/arduinoPlugin/package_index.json
After that delete this file and start the plugin again.

rlogiacco, 2016-04-09:
@jantje should we consider to add a warning regarding the computer being connected and network configured (proxy wise) before starting the download process?

wimjongman, 2016-04-09:
They just downloaded the plugin. I am fine with assuming that there is a network.

The Proxy is another thing.

jantje, 2016-04-09:
on the  last image you see the welcome page which is also downloaded. http://eclipse.baeyens.it/product/V3_0.shtml
Looking at it again it looks like this is some special network setup/issue (there is a redirect warning on the 2'nd and 3th picture)
So I think this is a duplicate for #393 except for the file is redirected on an intermediate network server which sounds like a middleman network attack.
In other words I thin @dhrupal93 his computer is (intentionally?) hacked.

As to the proxy
AFAIK eclipse proxy setting is by default: take the default network settings. As this process is started at the very beginning, eclipse proxy settings should not interfere.

dhrupal93, 2016-04-09:
Our campus has Proxy network, so when I installed with full eclipseArduino. it was not able to download the plugin after i opened for first time. Also it was asking for two .jason files (ESP and Arduino)

When I only opened eclipse (not eslipseArduino) . and then I dragged the plugin from market place. Now it works for eclipse with arduino.

rlogiacco, 2016-04-11:
@dhrupal93 as you correctly guessed this has to do with the network configuration. Toddler tries to download all the necessary files on first start and a proxy will prevent that download to complete nicely.

@jantje if they set the network properties and restart, would the initial download attempted again?

dhrupal93, 2016-04-12:
I tried that (open once, then set proxy and restart), but it was not working then also.

jantje, 2016-04-12:
At startup Toddler checks if he has "boards". If no boards are found the default are downloaded.
A proxy should not cause issues, a "redirection" does.
Again I think @dhrupal93 his system is infected.

@dhrupal93 
Can you try on another machine on the same campus?

jantje, 2016-05-07:
@dhrupal93 
Since there has not been any reaction in 25 days I'm closing this

closed 2016-05-07 by jantje