Sloeber/arduino-eclipse-plugin #415: Stuck at splash screen (Win10, 64bit)

funth1ngs, 2016-04-05:
Hi,

when I start V3 with eclipseArduinoIDE.exe (stable and nightly) i'm stuck at the splash screen after selecting the worskpace folder. The workspace/.metadata folder is created, but there is no .log file:
```
lock
.mylyn
.plugins
version.ini
```

With "eclipsec.exe -consoleLog -debug" I get the following (I'm also stuck at the spalsh screen):

```
Start VM: -Dosgi.requiredJavaVersion=1.8
-Dorg.eclipse.update.reconcile=false
-Xms40m
-Xmx512m
-Djava.class.path=C:\Users\Heiko\Arduino\eclipseArduino\\plugins/org.eclipse.equinox.launcher_1.3.100.v20150511-1540.jar
-os win32
-ws win32
-arch x86_64
-showsplash
-launcher C:\Users\Heiko\Arduino\eclipseArduino\eclipsec.exe
-name Eclipsec
--launcher.library C:\Users\Heiko\Arduino\eclipseArduino\\plugins/org.eclipse.equinox.launcher.win32.win32.x86_64_1.1.300.v20150602-1417\eclipse_1611.dll
-startup C:\Users\Heiko\Arduino\eclipseArduino\\plugins/org.eclipse.equinox.launcher_1.3.100.v20150511-1540.jar
--launcher.appendVmargs
-perspective it.baeyens.arduino.application.perspective
-consoleLog
-debug
-vm C:\Program Files\Java\jre1.8.0_77\bin\server\jvm.dll
-vmargs
-Dosgi.requiredJavaVersion=1.8
-Dorg.eclipse.update.reconcile=false
-Xms40m
-Xmx512m
-Djava.class.path=C:\Users\Heiko\Arduino\eclipseArduino\\plugins/org.eclipse.equinox.launcher_1.3.100.v20150511-1540.jar
Install location:
    file:/c:/Users/Heiko/Arduino/eclipseArduino/
Configuration file:
    file:/c:/Users/Heiko/Arduino/eclipseArduino/configuration/config.ini loaded
Configuration location:
    file:/c:/Users/Heiko/Arduino/eclipseArduino/configuration/
Framework located:
    file:/c:/Users/Heiko/Arduino/eclipseArduino/plugins/org.eclipse.osgi_3.10.102.v20160118-1700.jar
Loading extension: reference:file:org.eclipse.equinox.region_1.2.101.v20150831-1342.jar
        eclipse.properties not found
Loading extension: reference:file:org.eclipse.equinox.transforms.hook_1.1.0.v20131021-1933.jar
        eclipse.properties not found
Loading extension: reference:file:org.eclipse.osgi.compatibility.state_1.0.100.v20150402-1551.jar
        eclipse.properties not found
Framework classpath:
    file:/c:/Users/Heiko/Arduino/eclipseArduino/plugins/org.eclipse.osgi_3.10.102.v20160118-1700.jar
    file:/c:/Users/Heiko/Arduino/eclipseArduino/plugins/
    file:/c:/Users/Heiko/Arduino/eclipseArduino/plugins/org.eclipse.equinox.region_1.2.101.v20150831-1342.jar
    file:/c:/Users/Heiko/Arduino/eclipseArduino/plugins/org.eclipse.equinox.transforms.hook_1.1.0.v20131021-1933.jar
    file:/c:/Users/Heiko/Arduino/eclipseArduino/plugins/org.eclipse.osgi.compatibility.state_1.0.100.v20150402-1551.jar
Splash location:
    c:\Users\Heiko\Arduino\eclipseArduino\configuration\org.eclipse.equinox.launcher\it.baeyens.arduino.application_3.0.0.201604050209\splash.bmp
Debug options:
    file:/C:/Users/Heiko/Arduino/eclipseArduino/.options not found
Time to load bundles: 0
Starting application: 875
```

wimjongman, 2016-04-05:
Try to start with -clean

funth1ngs, 2016-04-05:
Hi Wim,

I already tried that, it didn't help.

jantje, 2016-04-05:
Win10 64bit is the os I currently work most with and I have never seen something like that.
Were there any problems with the download or extraction of the zip?

funth1ngs, 2016-04-05:
Nope, I used 7-Zip for extraction, everything worked fine...

wimjongman, 2016-04-05:
Some things to try.

How long did you wait?
Is the console responding if you also start with -console 
in the console, do "threads" to show the current running threads
Try to "run as administrator"
Where did you place the product?
Is there a folder "arduinoPlugin" created in the installation directory?
Does this directory get populated?
Are you using some network proxy? Can it connect to the net?
Can you make a stackdump?

funth1ngs, 2016-04-05:
> How long did you wait?

at least 15mins

> in the console, do "threads" to show the current running threads

```
ThreadGroupType:   Name:                 ParentGroup:   MaxP: Threads:
ThreadGroup        Equinox Event Admin   main             10   0/0
ThreadGroup        main                  system           10  14/14
ThreadType:        Name:                 ThreadGroup:   Prio:
Thread             Active Thread: Equino main              5
Thread             Attach Listener       system            5 [daemon]
EventManager$Event Bundle File Closer    main              6 [daemon]
CommonUtil$1Refere EMF Reference Cleaner main              5 [daemon]
Finalizer$Finalize Finalizer             system            8 [daemon]
EventManager$Event Framework Event Dispa main              5 [daemon]
Thread             Gogo shell            main              5
Reference$Referenc Reference Handler     system           10 [daemon]
Thread             Signal Dispatcher     system            9 [daemon]
EventManager$Event Start Level: Equinox  main              5 [daemon]
Thread             Thread-5              main              6
Thread             Thread-6              main              6
Worker             Worker-0              main              5
Worker             Worker-1              main              5
InternalWorker     Worker-JM             main              5
Executor           [ThreadPool Manager]  main              5 [daemon]
Thread             [Timer] - Main Queue  main              5 [daemon]
Thread             main                  main              6
```

> Try to "run as administrator"

same behavior

> Where did you place the product?

C:\Users\Heiko\Arduino\eclipseArduino

> Is there a folder "arduinoPlugin" created in the installation directory?

Nope

> Are you using some network proxy? Can it connect to the net?

No proxy connection

> Can you make a stackdump?

[stackdump.txt](https://github.com/jantje/arduino-eclipse-plugin/files/204661/stackdump.txt)

wimjongman, 2016-04-05:
If you do more stacktraces, at 1 minute intervals. Does it seem to move in the main thread?:

"main" #1 prio=6 os_prio=0 tid=0x000000000270e000 nid=0x560 in Object.wait() [0x000000000013e000]
   java.lang.Thread.State: TIMED_WAITING (on object monitor)
        at java.lang.Object.wait(Native Method)
        at it.baeyens.arduino.tools.ExternalCommandLauncher.launch(ExternalCommandLauncher.java:290)
        - locked <0x00000000f84ccc40> (a it.baeyens.arduino.tools.ExternalCommandLauncher)
        at it.baeyens.arduino.ui.Activator.testKnownIssues(Activator.java:80)
        at it.baeyens.arduino.ui.Activator.start(Activator.java:58)
        at org.eclipse.osgi.internal.framework.BundleContextImpl$3.run(BundleContextImpl.java:771)
        at org.eclipse.osgi.internal.framework.BundleContextImpl$3.run(BundleContextImpl.java:1)

funth1ngs, 2016-04-05:
I had it still running, and it showed the same output.

Doing the traces in the intervall also shows the same.

wimjongman, 2016-04-05:
```
	if (Platform.getOS().equals(Platform.OS_WIN32)) {
	    String shCommand = "sh --version"; //$NON-NLS-1$
	    boolean shFound = false;
	    ExternalCommandLauncher shCommandLauncher = new ExternalCommandLauncher(shCommand);
	    try {
---->		shFound = (shCommandLauncher.launch(null) == 0);
	    } catch (IOException e) {
		// nothing to do here
	    }
```

Here is where it is stuck. If you run this on the commandline what do you get?

c:> sh --version

funth1ngs, 2016-04-05:
It doesn't shows a version information:

> C:\Users\Heiko>sh --version
~ $

jantje, 2016-04-05:
What does it do when you use "where sh"

funth1ngs, 2016-04-05:
> C:\Users\Heiko>where sh
C:\Program Files\GNU ARM Eclipse\Build Tools\2.6-201507152002\bin\sh.exe

jantje, 2016-04-05:
You will need to get this one out of your path as sh in the path is causing problems.
Check for bash to "where bash"

wimjongman, 2016-04-05:
Right. Your version of sh just does not return. 

I don't think we should assume that **sh --version** returns something useful. In fact, it is very dangerous just executing some random program with the name "sh" on the path.

I think we should change our strategy here.

jantje, 2016-04-05:
I agree where is better. Though I don't know which windows versions support it

funth1ngs, 2016-04-05:
> Check for bash to "where bash"

Not found...

I removed the line from PATH, it's starting now!
Thank you both!

wimjongman, 2016-04-05:
The answer of

http://stackoverflow.com/questions/304319/is-there-an-equivalent-of-which-on-the-windows-command-line

says that it is ancient.

wimjongman, 2016-04-05:
Glad to help @funth1ngs. Consider a donation.

jantje, 2016-04-05:
As to my web loggings 2949 hits in January 2016 from XP :-(
The new and advanced statistics from February onwards no longer provide this info :-(
So unfortunately where seems not ancient enough.

wimjongman, 2016-04-05:
Microsoft stopped supporting XP 5 years ago and you want to continue supporting it?

jantje, 2016-04-05:
I am only interested in reducing time needed to support the plugin. I see there are still lots of people who use it. So the question is what will happen when we call "where sh" in xp or can we test "os==xp".

funth1ngs, 2016-04-05:
Why not use http://stackoverflow.com/a/304441

jantje, 2016-04-05:
@funth1ngs 
I fear this will need extra coding to parse the output. Now we simple test the return value of the whole call. Also the script does not cater for the local folder.

As @wimjongman implies simply using where and knowing this won't work in xp is probably the best way to go.

wimjongman, 2016-04-05:
Nice.

jantje, 2016-04-09:
@funth1ngs 
Can you confirm this works better now?

funth1ngs, 2016-04-10:
I get this message at startup:

```
we have found programs in the path that might conflict with our external builder.
The conflicting programs are sh.
The program might still function but if you get strange build errors you know where to look
```

After I hit OK, it's starting.

Cheers,
Heiko

jantje, 2016-04-10:
@funth1ngs 
txs for the feedback

closed 2016-08-01 by jantje