Sloeber/arduino-eclipse-plugin #421: Failure creating a new Arduino sketch project

Defossez, 2016-04-06:
Hi,

Using Eclipse_Mars.2 (with JAVA 1.8)
Installed the latest Arduino plugin from "http://www.baeyens.it/eclipse/V3".
        Tried even the nightly build install.
Everything goes well until creating a new "Arduino Sketch"
I can provide all information, project name, boards, select the board and I'm using (Adafruit Trinket Pro),
When I then hit [Finish] I get a popup box with following failure:
        Failed to find core reference: arduino:arduino 
                  I can select [Details >>] but that doesn't tell more.
                  Selecting [OK] brings up more popups with failures
        Failed to find variant reference: arduino:eightanaloginputs
Keep hitting [OK] finally ends in a new project.

The issues reported in the popups are things I didn't even select when creating the new project.
The only thing I did extra was pointing in the preference to a Adafruit json file to get the Adafruit stuff.

Thansk for any hints.

Defossez, 2016-04-06:
Ok, Did some extra tests.
Fresh Eclipse C/C++ package install + Arduino plugin V3
Create a new Arduino Sketch project can be done without any error/failure Selecting a Arduino Uno).
Add the AdaFruit json repository to the locations list (Preferences).
Create a new Arduine Sketch project using the previous selected Arduino Uno board works OK.
Create a new Arduino Sketch project selecting the Adafruit.txt file and Trinket Pro board result in a project with failures as previously described.

jantje, 2016-04-06:
>core arduino:arduino 

This does not work in V3. I actually don't see why you would still use this.
In other words Adafruit Trinket Pro is not supported right now.

Defossez, 2016-04-06:
I don't use "Core arduino:arduino".
I don't even know where it coming from.
I use one of the latest small board from Adafruit and that is the only thing that is selected in [Preferences - Platforms & Boards - adaruit - adafruit AVR - 1.4.6].
    When selecting the txt file and the board, teh two issues pop up.
Why? maybe God knows.

But since AdaFruit is not supported I will need to return to the Arduino IDE, pity.

Does this also mean that all the repositories (json) obtained via the "where do I find json URLs" link  in preferences--Locations are not supported?
When will AdaFruit be supported?

Thanks

jantje, 2016-04-06:
>I don't use "Core arduino:arduino".

The boards.txt file ...
>I don't even know where it coming from.

....referenced in the json file you use does
>Why? maybe God knows.

Don't be sarcastic. 

>But since AdaFruit is not supported I will need to return to the Arduino IDE, pity.

It is not "adaFruit is not supported". It is: The board you are using uses a feature that is currently not supported within the Arduino eclipse product.

>Does this also mean that all the repositories (json) obtained via the "where do I find json URLs" link in preferences--Locations are not supported?

No only the ones that use the redirection feature arduino::arduino.

>When will AdaFruit be supported?

I guess you do not fully grasp the nature of open source. No one working on this plugin gets paid (be it directly or indirectly) for his/her work on the open source you are using. The open source developer is investing it's own time and money to write the code and share it to help other people out. As such the open source developer decides what he will be doing and not you.
Why: because you are not a customer because you don't pay.
To convince an open source developer to do something you have a couple of do's and don't do.
Do's are
Show respect.
Show you invest time yourself by reading up and verifying your facts.
Support any way you can.

Don't do
moan
Act like a little kid "och my board is not supported now I have to ...."
Act as if you are all important (Only God knows)
Request solutions, answers, ....
Be as vague and misty as possible "It doesn't work"

In other words: use another tool or get your shoulders under it to get it.
I don't know your skills so I can't tell you can help to get this implemented. But some ideas:
If you are rich: Put 10 000 euro on the table to get this fixed.
If you can code: Do it yourself
If you can negotiate: 
  --Convince Adafruit to do the coding
   --Do crowdfunding to get the funds to do it
.......
And many more options.

No Thanks

gijs69, 2016-05-13:
Hi Jantje, I run into the same problem when I try to add the MySensors SenseBender Micro boards.

I guess this function needs some work:
setTheEnvironmentVariablesRedirectToOtherVendors

Am I right? 

In the comments in this function it seems that you already tried to fix this, but that it is rather complicated.

Any idea if you get to this in the near future?

jantje, 2016-05-13:
>Any idea if you get to this in the near future? 

Not without sponsors as I don't need/use it.
It was pretty complicated before the boardsmanager and when it finally was implemented the boardsmanager changed the whole thing.

gijs69, 2016-05-14:
That's a pity. Since it is just for hobby I won't be the sponsor :). 

Regards

Verstuurd vanaf mijn iPhone

universpharmacie, 2016-05-14:
gijs69, that's not a reason. I'm using eclipseArduino for now to develop a non-commercial project, which has nothing to do with my day job or with earning money. But I've become a sponsor, though, because jantje's (and the other contributor's) effort is worth it.

gijs69, 2016-05-14:
True. But at the moment I am just experiencing a bit. I should better have stated, I just started with some small Arduino projects and I am not yet sure whether I will be doing serious development projects that need a sophisticated IDE. 

Verstuurd vanaf mijn iPhone

jantje, 2016-05-15:
Note that there is a work around by simply copying the core arduino folder to your package and fiddling around with the platform.txt and boards.txt.

DamienMiras, 2016-05-28:
I have got the same error with a board with the "arduino:arduino" defined in the board.txt. The board is from sparkfun "arduion pro micro". I managed to make it works and I think the method I follow would works for many unsuported board.

the complete work arround involve the following step :
1. locate the board.txt in my case it is there eclipseArduino\arduinoPlugin\packages\SparkFun\hardware\avr\1.1.5
2. find "arduino:arduino" and replace it by "arduino"
3. locate platform.txt in the same folder, verify if there an "# AVR compile variables" if not copy that section from the plateform.txt found in the platform.txt of the uno  here : eclipseArduino\arduinoPlugin\packages\arduino\hardware\avr\1.6.5
4. copy the cores folder from eclipseArduino\arduinoPlugin\packages\arduino\hardware\avr\1.6.5 to your specific package folder, in my case it is eclipseArduino\arduinoPlugin\packages\SparkFun\hardware\avr\1.1.5
5. hit the compile button and enjoy that amazing plugin.

Please keep in mind that when you give a solution for someone, actually you give it not only for that person but for many other people out there. Thanks Jantje for that amazing works. I'am in the middle of the night, so I havent any time right now to read the source and help to solve that issue by a contribution. I use a lot of eclipse plugin and I can see NPE in the log stack trace. That's the first time I see a plugin where the exception are correctly handled and I must thank you for that. If I can't give some of my time for that project i would pay you a bear with a donation. thanks much

DamienMiras, 2016-05-28:
oh and I forgot, don't reload the json anymore because it will download and erase your work arround.

jantje, 2016-05-28:
@DamienMiras 
Thank you for documenting the work around in more detail.

flowh, 2016-06-01:
Very good work around, thank you very much. It solves my issue with "pro micro board". 
I am pretty knew with Eclipse. What does "don't reload json" mean?
I still have an error but it is linked with upload com port (it proposes 12 instead of 11??). I hope it is not linked with the previous issue.

Thank you again.

jantje, 2016-06-23:
I made a fix for this (referenced it wrongly to #423)
Should be available in the nightly in 6 hours.

DamienMiras, 2016-06-24:
thanks a lot, I do not had time  to write about the work arround indeed...

jantje, 2016-06-24:
@DamienMiras 
don't worry. Did you have time to test it?

DamienMiras, 2016-07-05:
probably in august yep i'm a bit buzy unfortunatly

jantje, 2016-07-06:
No problem :-)

closed 2016-08-01 by jantje