Sloeber/arduino-eclipse-plugin #424: Importing an ArduinoIDE project with the New Sketch Wizard and using same sketch-name, failed.

jrobert-gh, 2016-04-07:
I followed the procedure you posted in this [Arduino Stack Exchange answer](http://arduino.stackexchange.com/a/21022/313) but I was not able to create a new Eclipse project from an existing ArduinoIDE sketch.

- I created a sketch with the ArduinoIDE.
- In Eclipse I started the New Sketch Wizard and gave the new sketch the same name as I'd used in the ArduinoIDE. 
- What I expected: A new project containing the code I had already written.
- What actually happened: The original code was replaced by the pro-forma empty sketch.

jantje, 2016-04-08:
I'll have a look at this.

jantje, 2016-04-13:
I can confirm that the default files are created.
I'll see what is easiest: A no file project or a test to not replace when copying files.
What is your preference?

jrobert-gh, 2016-04-13:
I think "test to don't replace" is less error prone, by not depending on the user to make a correct selection of project in order to not lose his code. But either would be welcome!

jantje, 2016-04-13:
It seemed pretty simple. a fix should be available tomorrow on the nightly.

jrobert-gh, 2016-04-13:
Excellent - Thanks, Jantje.

jantje, 2016-04-14:
Can you confirm this works at your location?

jrobert-gh, 2016-04-14:
Yes, it did create an Eclipse project that included my existing files without damage.
The only hitch I encountered was that my source code had #included some .h files that Eclipse couldn't find and aren't needed. I deleted those #includes from my source code but the .ino.cpp file wasn't updated to delete them as well; I had to manually edit them out.

Thanks, Jantje.

jantje, 2016-04-14:
txs

closed 2016-08-01 by jantje