Sloeber/arduino-eclipse-plugin #428: Error Uploading on Arduino M0 Pro board

supiash1, 2016-04-09:
Hi,

Thanks for the plugin development for eclipse which was what exactly i had been trying to find when i failed to make things work with Eclipse+OpenOCD+Zylin Embedded Debugger.

I have issue in uploading the error is as below

Info : only one transport option; autoselect 'swd'
adapter speed: 500 kHz
adapter_nsrst_delay: 100
cortex_m reset_config sysresetreq
Info : CMSIS-DAP: SWD  Supported
Info : CMSIS-DAP: JTAG Supported
Info : CMSIS-DAP: Interface Initialised (SWD)
Info : CMSIS-DAP: FW Version = 01.1F.0118
Info : SWCLK/TCK = 1 SWDIO/TMS = 1 TDI = 1 TDO = 1 nTRST = 0 nRESET = 1
Info : CMSIS-DAP: Interface ready
Info : clock speed 500 kHz
Info : SWD IDCODE 0x0bc11477
Info : at91samd21g18.cpu: hardware has 4 breakpoints, 2 watchpoints
target state: halted
target halted due to debug-request, current mode: Thread 
xPSR: 0x81000000 pc: 0x000028f4 msp: 0x20002c00
** Programming Started **
auto erase enabled
Info : SAMD MCU: SAMD21G18A (256KB Flash, 32KB RAM)
Warn : Adding extra erase range, 00000000 to 0x00001fff
Error: SAMD: NVM lock error
Error: Failed to erase row containing 00002000
Error: error writing to flash at address 0x00000000 at offset 0x00002000
Runtime Error: embedded:startup.tcl:454: ** Programming Failed **
in procedure 'program' 
in procedure 'program_error' called at file "embedded:startup.tcl", line 510
at file "embedded:startup.tcl", line 454
0.9.0-arduino/bin/openocd.exe finished
upload done

I had faced the similar error when using Zylin Embedded debugger but when i have used the Arduino IDE from arduino.org it works. So I would think it would probably some change required at config. Any help is appreciated

Thanks

jantje, 2016-04-10:
I decided to no longer respond to github issues -and other questions- due to a lack of support.
As a patron you can post a message on the patreon forum stating you support a github issue. This allows me to support people who support me.
https://www.patreon.com/posts/3700469

That being said, I'm willing to help you to try to find out what is at hand.
First things first.  This plugin grew from arduino.cc, I do not know what arduino.org is doing and I actually don't want to know.
As you have a IDE that does the upload correctly do the upload in that tool with verbose mode on.
Copy the command and output of the command in a text editor.
Do the upload in the arduino eclipse plugin and copy the console content in another (instance) of a text editor.
Now you can compare the 2 texts.
The crucial thing to check first is the "upload command" If that is different a tweek in the platform.txt or text.txt can help you out.
If the "upload command" is 100% the same (functionally that is) there are some actions that need to be taken to make the board "upload ready" (like the reset for arduino boards). This is a harder case as you will have to find out what needs to be done and then whether there is already support in the plugin to do this.

Good luck
Jantje

jantje, 2016-09-03:
Closing due to no sponsor found

closed 2016-09-03 by jantje