Sloeber/arduino-eclipse-plugin #429: V3.0 To update the libray jason one needs to delete the json file

jantje, 2016-04-10:
The gui does not allow to upgrade to the latest library json file provided by arduino.

Work around:
When your eclipse instance is not running, delete the file [eclipse install]/arduinoPlugin/library_index.json
Restart Arduino eclipse plugin.

rlogiacco, 2016-04-10:
I believe this is a good candidate for an hotfix :smile: 

What if we check the ETag header and overwrite the saved file whenever necessary? Today ETag is `5708cf78-114521` and I've submitted a new library it's not yet in there.
We can compare the header value once the library gets into the index and if the ETag value is changed (as per HTTP specs) I believe we can rely on that to determine if the file needs to be replaced.

jantje, 2016-04-11:
I'm ok doing a hotfix for this one as an "exercise". For me there are 2 real reasons for a hotfix: Patron request or expectation to avoid extensive work (read the issue is going to produce many questions)

Your other part is a change request that should live in its own issue.
I'm a bit scared of using ETag due to of #393. Anyway #393 should be in the test case.

rlogiacco, 2016-04-11:
Does #393 also apply to libraries? I thought it would not...

jantje, 2016-04-12:
#393 has got to do with the download of the json file. This is exact the same code for libraries json as the boards json

rlogiacco, 2016-04-13:
May be a simpler solution is better: what if we check the file timestamp and perform an update let's say every 24h? We can also add a button at the end of the table/tree to force an immediate refresh...

Would that be a better solution?

jantje, 2016-04-13:
The problem I see is this: if the owner of the json decides to use the redirection feature (or any other feature which the plugin currently doesn't support) the plugin will become partly useless because the json ( in casu the library file) is corrupt. 
Any automated update will result in all users having the same issue in a relative small time-frame. This will create a huge pressure to fix this "soon" (or in other words create a hot-fix).

So a good solution would be to add "upgrade code" which can roll-back in case the upgrade fails. 
The current "upgrade code" is a "delete and add" which does not support roll-back.
Once that is done I can feel comfortable with a "auto upgrade" of the json files

closed 2016-08-01 by jantje