Sloeber/arduino-eclipse-plugin #433: can't find iostream library

mdilger, 2016-04-14:
Hi Guys,

It could be me, but I am struggling to solve an issue.  The PubSubClient library is trying to call the iostream library and I am unable solve the issue. At this stage I am just rying to run a sample application that is provided withe library.

I assume this is not an issue with the Arduino IDE...   but am unable to confirm this as I do not have it loaded.

Following is a log from the last compile attempt:

### 20:06:15 **** Incremental Build of configuration Release for project MQTT_4 ****
"C:\\Eclipse\\eclipseArduino\\arduinoPlugin\\tools\\make\\make" all 
'Building file: C:/Eclipse/eclipseArduino/arduinoPlugin/packages/arduino/hardware/avr/1.6.9/libraries/SPI/SPI.cpp'
'Starting C++ compile'
"C:\Eclipse\eclipseArduino\arduinoPlugin\tools\arduino\avr-gcc\4.8.1-arduino5/bin/avr-g++" -c -g -Os -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10606 -DARDUINO_AVR_ETHERNET -DARDUINO_ARCH_AVR     -I"C:\Eclipse\eclipseArduino\arduinoPlugin\packages\arduino\hardware\avr\1.6.9\cores\arduino" -I"C:\Eclipse\eclipseArduino\arduinoPlugin\packages\arduino\hardware\avr\1.6.9\variants\ethernet" -I"C:\Eclipse\eclipseArduino\arduinoPlugin\packages\arduino\hardware\avr\1.6.9\libraries\SPI" -I"C:\Eclipse\eclipseArduino\arduinoPlugin\libraries\PubSubClient\2.6.0" -I"C:\Eclipse\eclipseArduino\arduinoPlugin\libraries\PubSubClient\2.6.0\src" -I"C:\Eclipse\eclipseArduino\arduinoPlugin\libraries\Ethernet\1.1.1" -I"C:\Eclipse\eclipseArduino\arduinoPlugin\libraries\Ethernet\1.1.1\src" -I"C:\Eclipse\eclipseArduino\arduinoPlugin\packages\arduino\hardware\avr\1.6.9\libraries\SoftwareSerial" -MMD -MP -MF"libraries/SPI/SPI.cpp.d" -MT"libraries/SPI/SPI.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "C:/Eclipse/eclipseArduino/arduinoPlugin/packages/arduino/hardware/avr/1.6.9/libraries/SPI/SPI.cpp"  -o  "libraries/SPI/SPI.cpp.o"   -Wall
'Finished building: C:/Eclipse/eclipseArduino/arduinoPlugin/packages/arduino/hardware/avr/1.6.9/libraries/SPI/SPI.cpp'
' '
'Building file: C:/Eclipse/eclipseArduino/arduinoPlugin/libraries/PubSubClient/2.6.0/tests/src/lib/BDDTest.cpp'
'Starting C++ compile'
"C:\Eclipse\eclipseArduino\arduinoPlugin\tools\arduino\avr-gcc\4.8.1-arduino5/bin/avr-g++" -c -g -Os -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10606 -DARDUINO_AVR_ETHERNET -DARDUINO_ARCH_AVR     -I"C:\Eclipse\eclipseArduino\arduinoPlugin\packages\arduino\hardware\avr\1.6.9\cores\arduino" -I"C:\Eclipse\eclipseArduino\arduinoPlugin\packages\arduino\hardware\avr\1.6.9\variants\ethernet" -I"C:\Eclipse\eclipseArduino\arduinoPlugin\packages\arduino\hardware\avr\1.6.9\libraries\SPI" -I"C:\Eclipse\eclipseArduino\arduinoPlugin\libraries\PubSubClient\2.6.0" -I"C:\Eclipse\eclipseArduino\arduinoPlugin\libraries\PubSubClient\2.6.0\src" -I"C:\Eclipse\eclipseArduino\arduinoPlugin\libraries\Ethernet\1.1.1" -I"C:\Eclipse\eclipseArduino\arduinoPlugin\libraries\Ethernet\1.1.1\src" -I"C:\Eclipse\eclipseArduino\arduinoPlugin\packages\arduino\hardware\avr\1.6.9\libraries\SoftwareSerial" -MMD -MP -MF"libraries/PubSubClient/tests/src/lib/BDDTest.cpp.d" -MT"libraries/PubSubClient/tests/src/lib/BDDTest.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "C:/Eclipse/eclipseArduino/arduinoPlugin/libraries/PubSubClient/2.6.0/tests/src/lib/BDDTest.cpp"  -o  "libraries/PubSubClient/tests/src/lib/BDDTest.cpp.o"   -Wall
In file included from C:/Eclipse/eclipseArduino/arduinoPlugin/libraries/PubSubClient/2.6.0/tests/src/lib/BDDTest.cpp:2:0:
C:/Eclipse/eclipseArduino/arduinoPlugin/libraries/PubSubClient/2.6.0/tests/src/lib/trace.h:3:20: fatal error: iostream: No such file or directory
 #include <iostream>
                    ^
compilation terminated.
libraries/PubSubClient/tests/src/lib/subdir.mk:30: recipe for target 'libraries/PubSubClient/tests/src/lib/BDDTest.cpp.o' failed
make: *** [libraries/PubSubClient/tests/src/lib/BDDTest.cpp.o] Error 1

20:06:16 Build Finished (took 656ms)

Sorry but after searching for a solution I have come up with nothing.  Any help or suggestions would be appreciated.

Cheers,
Mike

jantje, 2016-04-14:
@mdilger 
First of all: Thanks for supporting us on patreon.
I had a look and the problem is related to a subfolder in the library that is called "tests".
`C:/Eclipse/eclipseArduino/arduinoPlugin/libraries/PubSubClient/2.6.0/tests`
I ignore the subfolder "test" but not "tests".
I'll change the ignoring pattern to "test*" which covers test and tests
There are 3 things you can do to workaround.
1) rename the tests folder on your disk to test. This will be ok untill you upgrade or reinstall the library.
2) In you project go to `libraries/PubSubClient/2.6.0/tests` right click it and select "resource configurations->ecluse from build->select all->ok". This will only fix the problem for this project.
3) The last one and most difficult is to change the filter to test*. In project properties->C/C++general->paths and symbols->tab:source location->open the source folder for your project->click the filter->edit filter->select libraries/**/test/** change it to libraries/**/test*/**. This will again only change it for this particular project.

Best regards
Jantje

rlogiacco, 2016-04-14:
@jantje I'm not sure that exclusion is going to work nicely in every case.... What if I have a library which defines, let's say a class `tester/ResistenceMeter`?

jantje, 2016-04-14:
@rlogiacco I'll test what that gives.
Note that due to mark down there are stars missing in the filters

jantje, 2016-04-14:
I think I found some nice filters :-)

jantje, 2016-04-14:
@mdilger 
Can you test this with the build of tomorrow?

mdilger, 2016-04-14:
@jante 
No problem, I will test and confirm.  Roughly when do you expect the new build will be available to download?

jantje, 2016-04-14:
6 hours from now

mdilger, 2016-04-15:
@jante

Set up a fresh sketch then tested and all OK.  Thank you very much for the quick attention to this issue.

Best regards,
Mike

jantje, 2016-04-15:
This needs to be open as it is not yet fixed in a release.

closed 2016-08-01 by jantje