Sloeber/arduino-eclipse-plugin #434: How to sync manual reset with upload?

jrobert-gh, 2016-04-15:
With a device having no auto-reset ability and an Optiboot loader, manually syncing the reset button with upload is hit-or-miss, and frequently, miss. It seems that the "right" time to release the device's reset button varies from the first or first few successful uploads in an Eclipse session, to subsequent uploads. Syncing to the "DTR" messages isn't very successful. Waiting for "Launching ...avrdude" isn't either, but it is somewhat more likely to succeed once there have been a couple of successful uploads during that session.

Can a "GO" message of some type be displayed to help the user know when to release the reset button?  I would have guessed the time for this to happen is during avrdude's run and therefor out of Eclipse's control, seeing how variable it is, I'm not convinced of that.

rlogiacco, 2016-04-16:
I'm far from being an expert on this, but I believe it depends on many factors, including board type and bootloader code so printing a *>> GO RESET NOW<<* string in the console output is not going to make any difference.
I'll let more experienced people think about this, but my vote about this is :-1:

jantje, 2016-04-16:
As @rlogiacco mentioned adding a go will not work.
I tried some alternatives but none showed signs of a potential solution.
so I vote :-1:

jrobert-gh, 2016-04-17:
Thank you for taking a look at it.

closed 2016-04-17 by jrobert-gh

jrobert-gh, 2017-06-09:
I have found the answer my own question:

- Hold the reset button on your board-duino.
- Click the Upload button in the EclipseArduino toolbar
- Watch the console window for these messages:
```
Starting upload
using arduino loader
Starting reset using DTR toggle process
Toggling DTR
Continuing to use/dev/cu.usbserial-AI03HPT0
Ending reset

Launching/Applications/sloeber.app/Contents/Eclipse//arduinoPlugin/packages/arduino/... 

Output:
```
- When this message appears:
`Launching/Applications/sloeber.app/....`
, count "One thousand one more" and release the reset button.

jantje, 2017-06-09:
I'm happy you got it working.