Sloeber/arduino-eclipse-plugin #437: Stable ArduinoEclipse + Maple Mini + UBUNTU 14.04   compile Error

Tom225, 2016-04-19:
Hello,
I have problem with compiling basic example Blink.ino for maple mini under Ubuntu 14.04 LTS. I am new in C++ programming. First,I tested this example Blink.ino in Arduino IDE under Ubuntu 14.04 and it works fine. But I need work in Eclipse. 

Example:
// the setup function runs once when you press reset or power the board
void setup() {
  // initialize digital pin 13 as an output.
  pinMode(33, OUTPUT);
}
// the loop function runs over and over again forever
void loop() {
  digitalWrite(33, HIGH);   // turn the LED on (HIGH is the voltage level)
  delay(1000);              // wait for a second
  digitalWrite(33, LOW);    // turn the LED off by making the voltage LOW
  delay(1000);              // wait for a second
}

I have had installed:
1/ Arduino IDE – version 1.6.5
2/ Arduino_STM32 in /opt/arduino-1.6.5-r5/hardware
3/ stable version of ArduinoEclipse – it is moved into opt/eclipseArduino
from: http://eclipse.baeyens.it/stable-linux.html
4/ I set path in EclipseArduinoIDE/Window/Preferences/Arduino:
Private Library path: /opt/arduino-1.6.5-r5/hardware/Arduino_STM32/STM32F1/libraries
Private Hardware path: /opt/arduino-1.6.5-r5/hardware/

5/ some changes according to (viewtopic.php?t=314):
******************************************************************************************
add lines:
file: Arduino_STM32/STM32F1/platform.txt
build.use_archiver=false
build.system.path={runtime.platform.path}/system
bootloader.tool=maple_upload
change lines:
tools.maple_upload.path.linux={runtime.platform.path}/../tools/linux
compiler.path={runtime.tools.arm-none-eabi-gcc.path}/bin/

file: Arduino_STM32/tools/linux/maple_upload
before ${DFU_UTIL} line:
DIR=$( cd "$( dirname "${BASH_SOURCE[0]}" )" && pwd )
$DIR/reset.py
sleep 1
change line:
${DFU_UTIL} -d ${usbID} -a ${altID} -D ${binfile} -R
***********************************************************************************************

Then I compiled example and there was error:

16:30:08 **** Incremental Build of configuration Release for project test ****
make all
Building file: ../.ino.cpp
Starting C++ compile
"/bin/arm-none-eabi-g++" -c -g -Os -DDEBUG_LEVEL=DEBUG_NONE -MMD -ffunction-sections -fdata-sections -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -DBOARD_maple_mini -DVECT_TAB_ADDR=0x8005000 -DERROR_LED_PORT=GPIOB -DERROR_LED_PIN=1 -mcpu=cortex-m3 -DF_CPU=72000000L -DARDUINO=10606 -DARDUINO_MAPLE_MINI -DARDUINO_ARCH_STM32F1 -DMCU_STM32F103CB -DSERIAL_USB -mthumb -march=armv7-m -D__STM32F1__ -DMCU_STM32F103CB -DSERIAL_USB -mthumb -march=armv7-m -D__STM32F1__ "-I/opt/arduino-1.6.5-r5/hardware/Arduino_STM32/STM32F1/system/libmaple" "-I/opt/arduino-1.6.5-r5/hardware/Arduino_STM32/STM32F1/system/libmaple/include" "-I/opt/arduino-1.6.5-r5/hardware/Arduino_STM32/STM32F1/system/libmaple/stm32f1/include" "-I/opt/arduino-1.6.5-r5/hardware/Arduino_STM32/STM32F1/system/libmaple/usb/stm32f1" "-I/opt/arduino-1.6.5-r5/hardware/Arduino_STM32/STM32F1/system/libmaple/usb/usb_lib" -I"/opt/arduino-1.6.5-r5/hardware/Arduino_STM32/STM32F1/cores/maple" -I"/opt/arduino-1.6.5-r5/hardware/Arduino_STM32/STM32F1/variants/maple_mini" -MMD -MP -MF".ino.cpp.d" -MT".ino.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "../.ino.cpp" -o ".ino.cpp.o" -Wall
/bin/sh: 1: /bin/arm-none-eabi-g++: not found
make: *** [.ino.cpp.o] Error 127

16:30:09 Build Finished (took 148ms)

I tried to find the file „ /bin/arm-none-eabi-g++“ and here is the place where I found it:

/home/core2pc/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin

I found thread about similar problem:  https://github.com/arduino/Arduino/issues/2982
*********************************************************************************************************************
TEXT:
The arduino eclipse plugin does not work with arduino IDE 1.6.2 and 1.6.3
The reason is the definition of compiler.path in platform.txt
In Arduino IDE 1.6.1 in avr it is defined as
compiler.path={runtime.ide.path}/hardware/tools/avr/bin/
In Arduino IDE 1.6.2 and 1.6.3 in avr it is defined as
compiler.path={runtime.tools.avr-gcc.path}/bin/
The problem is that runtime.tools.avr-gcc.path is not defined in the platform.txt so it adds another unneeded and probably undocumented environment variable.
There is a very similar situation with sam
END OF TEXT
********************************************************************************************************************
I don´t know how this problem was solved. I tried to add one missing environment variable: A.RUNTIME.TOOLS.ARM-NONE-EABI-GCC.PATH, but it was not saved. 

Could anybody help me? 
I think that Mr. Baeyens will help me. I need upload firmware into Maple Mini under Ubuntu.

Thanks.

Tomas

jantje, 2016-04-19:
I decided to no longer respond to github issues -and other questions- due to a lack of support.
As a patron you can post a message on the patreon forum stating you support a github issue. This allows me to support people who support me.
https://www.patreon.com/posts/3700469

Tom225, 2016-04-20:
Excuse me, English is not my native language. I used incorrect words. I am sorry. I did not want to injure anybody. Now I know, that the text could sound unseemly. But it is a misunderstanding.  I am sorry.  I study English only one year. When I was writing the text,  I was in hurry. I was in time pressure. Excuse me, please. 
Tomas

jantje, 2016-04-20:
@Tom225 
My previous response is a standard response, so there is no reason to feel sorry.

jantje, 2016-09-03:
There seems to be no json file for maple. So maple is not supported

closed 2016-09-03 by jantje