Sloeber/arduino-eclipse-plugin #438: Use users folder (libraries/hardware) relative path when possible.

far5893, 2016-04-20:
In" Arduino Settings for this workspace",
file dialog does not permit to choose path (library/hardware) , relative to HOME folder or workspace.
Instead use absolute path.

In conjunction with CVS (or simply workspace folder copy) there are a lot of absolute references to change in projects config files.

jantje, 2016-04-20:
duplicateof #34

closed 2016-04-20 by jantje

far5893, 2016-04-20:
#34 three years ago......bad news!!

wimjongman, 2016-04-20:
> #34 three years ago......bad news!!

The good news is that we are accepting quality patches!!

jantje, 2016-04-20:
I also posted the 6th of jan (this year) that I think I found a solution to fix it.
But nobody responded.

far5893, 2016-04-21:
I found broken link so i not consider it.

jantje, 2016-04-21:
>I found broken link so i not consider it.

I would consider reporting the broken link as constructive.
(Feel free the fill in the rest)
No thanks