Sloeber/arduino-eclipse-plugin #439: V3 Support of OTA for esp8266/Arduino

krzychb, 2016-04-22:
Uploading of esp8266 from [Arduino IDE 1.6.8 using OTA](https://github.com/esp8266/Arduino/blob/master/doc/ota_updates/readme.md#arduino-ide) consists of two steps:
1.	Select OTA port
2.	Press upload

[image]

This process is easy and simple as uploading with a serial port thanks to [ArduinoOTA](https://github.com/esp8266/Arduino/tree/master/libraries/ArduinoOTA) library and new functionality of Arduino IDE implemented with [PR-4107](https://github.com/arduino/Arduino/pull/4107).

Arduino Eclipse Plugin V3 already has functionality of detecting and selection of OTA port. 

[image]

Unfortunately after attempting to upload using esp8266 OTA port, unlike in Arduino IDE, the following window is displayed:

[image]

•	Does V3 support OTA uploads for esp8266 with ArduinoOTA library like Arduino IDE does?
•	If not, does it provide any other method of OTA uploads for esp8266?

Congratulations on developing really easy to use product. 
Integration with Arduino boards and libraries is really seamless. 
I was able to make it up and running as quickly as Arduino IDE while getting incomparable wealth of features of Eclipse :+1:

Krzysztof

jantje, 2016-04-22:
esp8266 is not my strongest area but I think this is a duplicate of #333
FYI currently the plugin assumes that you have a yun when you select an Ip address (found by the bonjour service -also called zero config-) in the port field.
From the link you gave it looks like the esp8266 wireless upload follows a different protocol.

krzychb, 2016-04-22:
Hi @jantje,

I was reading #333 some time ago and concluded it may take a lot of time and effort to get it implemented strictly the Arduino IDE way.

There may be a quicker solution.

To do OTA for esp8266 that has IP of 192.168.1.112, I need to run command like:
```
python.exe (…)eclipseArduino\arduinoPlugin\packages\esp8266\hardware\esp8266\2.1.0/tools/espota.py -i 192.168.1.112 -p 8266 --auth= -f  (…)eclipseArduino\workspace\BasicOTA/Release/BasicOTA.bin
```
Could you point me to a configuration file of Arduino Eclipse Plugin V3 that I can update to conditionally execute such command?

- The condition would be selection of OTA under `Upload Using:`.
- I would need to pass to this command IP address selected under `Port `.

I hope it is doable to convince plugin to assume esp8266 instead of yun when I select an Ip address **and** OTA :smile:

Krzysztof

[image]

jantje, 2016-04-22:
I had a look at this but apart from the forced yun upload there seems to be another bug.
When using OTA with the board nodemcuv2 the upload tool should be 
```
nodemcuv2.menu.UploadTool.espota=OTA
nodemcuv2.menu.UploadTool.espota.upload.tool=espota
```
However in my test it is esptool.
I won't be looking soon to this unless a sponsor steps up at https://www.patreon.com/posts/3700469

krzychb, 2016-04-23:
Hi jantje,

Thank you for quick review.
I assumed it as an easy intermediate step to have any OTA support before  [PR-4107](https://github.com/arduino/Arduino/pull/4107) / #333 is implemented. As this is not the case I would not explore it.

Krzysztof

jantje, 2016-04-23:
Yes but the bug still needs to be looked at. So I am reopening this.

krzychb, 2016-04-25:
Hi @jantje,

If by bug you mean any issue with menu options listed [above](https://github.com/jantje/arduino-eclipse-plugin/issues/439#issuecomment-213578575), and you get funding to look into it, please consider the following:

After implementation of [PR-4107](https://github.com/arduino/Arduino/pull/4107) in Arduino IDE, these options become irrelevant and have been removed from [boards.txt](https://github.com/esp8266/Arduino/blob/master/boards.txt) of esp8266/Arduino in [PR-1747](https://github.com/esp8266/Arduino/pull/1747). The latest [esp8266/Arduino](https://github.com/esp8266/Arduino) core [release 2.2.0](https://github.com/esp8266/Arduino/releases/tag/2.2.0) does not contain them at all.

Uploading of esp8266 is executed with the following script entries in [platform.txt](https://github.com/esp8266/Arduino/blob/master/platform.txt) file:

For serial upload:
```
tools.esptool.upload.pattern="{path}/{cmd}" {upload.verbose} -cd {upload.resetmethod} -cb {upload.speed} -cp "{serial.port}" -ca 0x00000 -cf "{build.path}/{build.project_name}.bin"
```

For network (OTA) upload:
```
tools.esptool.upload.network_pattern="{network_cmd}" "{runtime.platform.path}/tools/espota.py" -i "{serial.port}" -p "{network.port}" "--auth={network.password}" -f "{build.path}/{build.project_name}.bin"
```
Specifically to facilitate OTA upload above, esp8266 is advertising the following parameters using mDNS:
- ``` serial.port ``` IP address selected by user,
- ``` network.port ``` UDP port.

A sample of all advertised parameters is shown below:
[image]
[PR-4107](https://github.com/arduino/Arduino/pull/4107) implements reading of these parameters by Arduino IDE to pass them to OTA upload script entry discussed above.

To make esp8266 "OTA capable" it has to run [ArduinoOTA](https://github.com/esp8266/Arduino/tree/master/libraries/ArduinoOTA) routines. 

Esp8266 OTA functionality following this pattern is demonstrated by [BasicOTA.ino](https://github.com/esp8266/Arduino/tree/master/libraries/ArduinoOTA/examples/BasicOTA) sketch distributed in [esp8266 / Arduino](https://github.com/esp8266/Arduino) package among  examples.

Krzysztof

jantje, 2016-04-25:
I have a real difficulty to understand this whole OTA thing. But that is not related to the bug for which I want this issue to remain open.
Basically the "environment variables in eclipse are not set correctly given the menu selection made".

jantje, 2016-05-04:
this is sort of a racing condition

What happens is that the boards.txt is processed line by line. In the esp file the upload.tool is specified 3 times for the tested board.
Once as upload.tool and 2 times for the menu items upload protocol (serial and OTA).
Apparently the upload.tool was processed after the menu.
So one could say that this is a bug in the boards.txt file because upload.tool should not be specified.
However I catered for this situation.

Now when selecting ATO (with a com port) it will try to run python as expected.

krzychb, 2016-05-05:
@jantje 

I have tried latest nightly build [win32.2016-05-05_02-17-24.tar.gz](http://eclipse.baeyens.it/download/product/win32.2016-05-05_02-17-24.tar.gz) and still see a pop up asking for a login and password as described in first post above. Is https://github.com/jantje/arduino-eclipse-plugin/commit/7e68213a51b9e0ac66f7ad0c9d18789c4d195bb6 included in this nightly build?

What particular ```upload.tool ``` lines do you mean?

After implementation of [PR-4107](https://github.com/arduino/Arduino/pull/4107) in Arduino IDE, all menu entries like:
```
nodemcuv2.menu.UploadTool.esptool=Serial
nodemcuv2.menu.UploadTool.esptool.upload.tool=esptool
nodemcuv2.menu.UploadTool.esptool.upload.verbose=-vv
nodemcuv2.menu.UploadTool.espota=OTA
nodemcuv2.menu.UploadTool.espota.upload.tool=espota
```
become irrelevant and have been removed from [boards.txt](https://github.com/esp8266/Arduino/blob/master/boards.txt) of esp8266/Arduino in [PR-1747](https://github.com/esp8266/Arduino/pull/1747). The latest [esp8266/Arduino](https://github.com/esp8266/Arduino) core [release 2.2.0](https://github.com/esp8266/Arduino/releases/tag/2.2.0) that I see available in [win32.2016-05-05_02-17-24.tar.gz](http://eclipse.baeyens.it/download/product/win32.2016-05-05_02-17-24.tar.gz) does not contain them at all.

Krzysztof

[image]

jantje, 2016-05-05:
This fix should launch python if you select OTA and a standard com port
In my version I have following in the boards.txt

> nodemcuv2.upload.tool=esptool
> nodemcuv2.menu.UploadTool.esptool=Serial
> nodemcuv2.menu.UploadTool.esptool.upload.tool=esptool
> nodemcuv2.menu.UploadTool.esptool.upload.verbose=-vv
> nodemcuv2.menu.UploadTool.espota=OTA
> nodemcuv2.menu.UploadTool.espota.upload.tool=espota

It is the trouble caused by the first line that is fixed here.
In the case OTA is selected with the content above in the boards.txt; before the fix upload.tool was esptool after the fix upload.tool is espota.

The same can be achieved with modifying boards.txt to 

> nodemcuv2.menu.UploadTool.esptool=Serial
> nodemcuv2.menu.UploadTool.esptool.upload.tool=esptool
> nodemcuv2.menu.UploadTool.esptool.upload.verbose=-vv
> nodemcuv2.menu.UploadTool.espota=OTA
> nodemcuv2.menu.UploadTool.espota.upload.tool=espota

me-no-dev, 2016-05-06:
to get OTA to work, a new "GenericUploader" is needed to handle commands from the platform.txt, then when a network board is found, a few new properties are added to it that distinguish between which uploader should be selected (SSH or Generic). Nothing else should be required to get it to work (and I have it working in my fork)

jantje, 2016-05-06:
This bug originates from OTA upload but is no longer linked to it. Please discuss OTA in #333

krzychb, 2016-05-06:
Hi @jantje,

Your guidance https://github.com/jantje/arduino-eclipse-plugin/issues/333#issuecomment-217465385 made the trick and OTA works :smile: 
Linux/Windows users should have Avahi/Bonjour [installed](https://github.com/esp8266/Arduino/blob/master/doc/ota_updates/readme.md#application-example-1). 
Mac users are good to go by default.

Thank you for making OTA for ESP8266 available in V3!

Good luck with https://github.com/jantje/arduino-eclipse-plugin/issues/333 :+1:

Krzysztof

[image]

closed 2016-08-01 by jantje

cmburns, 2019-02-19:
Hi,

I am totally new to sloeber together with OTA. I run sloeber on my Windows 10 professional machine.
I have the BasicOTA running on my ESP8266. Then I changed the BasicOTA inside loop() to write something to serial. This changed sketch I wanted to use to test OTA.

What I am missing is the OTA port in the project properties. I only see the COM3.
Python 2.7.13 is installed.

What I further have to do to see the OTA-Port under the ports?

Thanks a lot!