Sloeber/arduino-eclipse-plugin #440: Compiling error with Genuino 101

dIoT, 2016-04-23:
Successfully created a project for Arduino 101 board using V3.0_win64.2016-03-31_12-43-53.tar.gz.  However, the following error occurs when trying to "Verify":  

> IDE version incompatible with Arduino 101. Please upgrade to version 1.6.7 or newer.

Below is what is printed at the console:

13:50:49 **** Incremental Build of configuration Release for project Erabliere ****
"C:\\Eclipse\\eclipseArduino\\arduinoPlugin\\tools\\make\\make" all 
'Building file: ../.ino.cpp'
'Starting C++ compile'
"C:\Eclipse\eclipseArduino\arduinoPlugin\tools\Intel\arc-elf32\1.6.4+1.0/bin/arc-elf32-g++" -c -mARCv2EM -mav2em -mlittle-endian -g -Os -Wall -fno-reorder-functions -fno-asynchronous-unwind-tables -fno-omit-frame-pointer -fno-defer-pop -Wno-unused-but-set-variable -Wno-main -ffreestanding -fno-stack-protector -mno-sdata -ffunction-sections -fdata-sections -fsigned-char -MMD -fno-rtti -fno-exceptions -D__ARDUINO_ARC__ -std=c++11 -DF_CPU=32000000L -DARDUINO=10606 -DARDUINO_ARC32_TOOLS -DARDUINO_ARCH_ARC32 -D__CPU_ARC__ -DCLOCK_SPEED=32 -DCONFIG_SOC_GPIO_32 -DCONFIG_SOC_GPIO_AON -DINFRA_MULTI_CPU_SUPPORT -DCFW_MULTI_CPU_SUPPORT -DHAS_SHARED_MEM "-IC:/Eclipse/eclipseArduino/arduinoPlugin/packages/Intel/hardware/arc32/1.0.5/system/libarc32_arduino101/common" "-IC:/Eclipse/eclipseArduino/arduinoPlugin/packages/Intel/hardware/arc32/1.0.5/system/libarc32_arduino101/drivers" "-IC:/Eclipse/eclipseArduino/arduinoPlugin/packages/Intel/hardware/arc32/1.0.5/system/libarc32_arduino101/bootcode" "-IC:/Eclipse/eclipseArduino/arduinoPlugin/packages/Intel/hardware/arc32/1.0.5/system/libarc32_arduino101/framework/include"    -I"C:\Eclipse\eclipseArduino\arduinoPlugin\packages\Intel\hardware\arc32\1.0.5\cores\arduino" -I"C:\Eclipse\eclipseArduino\arduinoPlugin\packages\Intel\hardware\arc32\1.0.5\variants\arduino_101" -I"C:\Eclipse\eclipseArduino\arduinoPlugin\packages\Intel\hardware\arc32\1.0.5\libraries\CurieBLE" -I"C:\Eclipse\eclipseArduino\arduinoPlugin\packages\Intel\hardware\arc32\1.0.5\libraries\CurieBLE\src" -MMD -MP -MF".ino.cpp.d" -MT".ino.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "../.ino.cpp"  -o  ".ino.cpp.o"   -Wall
In file included from ../.ino.cpp:7:0:
C:\Eclipse\eclipseArduino\arduinoPlugin\packages\Intel\hardware\arc32\1.0.5\cores\arduino/Arduino.h:33:2: error: #error IDE version incompatible with Arduino 101. Please upgrade to version 1.6.7 or newer.
 #error IDE version incompatible with Arduino 101. Please upgrade to version 1.6.7 or newer.
  ^
subdir.mk:24: recipe for target '.ino.cpp.o' failed
make: *** [.ino.cpp.o] Error 1

13:50:50 Build Finished (took 711ms)

jantje, 2016-04-23:
I decided to no longer respond to github issues -and other questions- due to a lack of support.
As a patron you can post a message on the patreon forum stating you support a github issue. This allows me to support people who support me (starting from as low as 1$ a month exclusive VAT).
https://www.patreon.com/posts/3700469

jantje, 2016-06-20:
Thanks to Ruud to support this issue.
This line should actually be a warning that you need to use arduino IDE 1.6.7 or newer.
However the arduino core developers decided to make this an error. 
As the code is compatible with the arduino nightly you can ignore this error.
Work arounds are: 
1 delete the line with with # error (I had to add a space for the editor)
2 replace the # error with # warning

The only thing I can do right now is upgrade the "Arduino IDE" version number in the plugin

jantje, 2016-06-20:
should be fixed tomorrow

jantje, 2016-06-22:
This is fixed here. Can you confirm it is fixed at your side to?

rmerriam, 2016-07-04:
I am not seeing the fix. I just ran an update. Here is what is in the log:

A bundle is already installed with the name "it.baeyens.arduino.core" and version "3.0.0.201603312133"

A bundle is already installed with the name "it.baeyens.arduino.application" and version "3.0.0.201603312133"

Rud

jantje, 2016-07-05:
You are still on the stable. This fix is only available in the nightly.
Are you using the nightly update site? http://eclipse.baeyens.it/nightly

rmerriam, 2016-07-05:
You're correct. I'm on the stable build. I generally avoid nightly to 
avoid the possible interruptions.

I changed the #ERROR to #WARNING to move forward. Turns out I don't want 
to use the A-101 for the current activity.

-73 -
*Rud Merriam K5RUD*
/Mystic Lake Software/ <http://mysticlakesoftware.com/>
/Hackaday Contributor/ <http://hackaday.com/>

jantje, 2016-07-05:
3.1 is expected 1th of august. Note that there is also issue #373 which has a work around and a not yet confirmed in sloeber nightly

closed 2016-08-01 by jantje