Sloeber/arduino-eclipse-plugin #441: The .ino.cpp file is no longer updated during the built.

EdwinCroissant, 2016-04-25:
You’re doing an excellent job and I’m sorry to bother you with the trivial problem I’m experiencing :) 
I’m on the nightly with Win10 and somewhere between April 12th and April 15th the .ino.cpp file is no longer updated during the built. My work around at the moment is to declare the functions before calling them. Probably I broke something but I don’t know what. Any hints will be appreciated.

jantje, 2016-04-25:
I checked here and it is indeed broken.
I have no clue what caused this.
Workaround is to remove the generated.ino.cpp do a clean and rebuild.
Not nice
Are you sure about the time frame?. I ask as I do not see changes in this time frame in this area.
Thanks for letting us know.

EdwinCroissant, 2016-04-25:
Thank you for checking this. Your work around is much better :) I checked the dates and found an updated .ino.ccp file dated April 14th and another one dated April 15th that was not updated. The difference is also in the file timestamp with the one from the .ino file after a build. I normally update eclipse before I start programming. But if I didn’t this could have started earlier. Good luck!

jantje, 2016-04-28:
I had another look at this and it works in my debugging environment but it does not work in my production environment :-(
I have no clue what is going on here.

EdwinCroissant, 2016-04-29:
I don’t know if this is related:

After the last update I was greeted with a java zero pointer exception.  In my current project the SD library attached itself and removing this library was not possible as the Finish button of the Add a library to the current project dialog stopped working.  I recovered by removing the offending entries in the linked resources and the include folders of the Arduino C++ and Arduino C compiler of the project properties. After that the java zero pointer exception was gone and I could build the project as normal.

Beside this your project rocks :)

Maybe this helps :)

jantje, 2016-04-29:
One of the latest changes added the programmers. This means the programmer is not filled in. That can cause the null pointer.
The standard upgrade procedure fixes that. The standard upgrade procedure is described here. http://eclipse.baeyens.it/how_to.shtml#/n
Once done the standard upgrade procedure and it still doesn't work we talk about regression or a bug.

i-copter, 2016-04-29:
Deleting the .ino.cpp file seemed to work for me.  

Mac OS X version nightly build 27-Apr-2016

Thanks.

EdwinCroissant, 2016-04-29:
Thank you jantje, forgot about that one, saved my day for the my other projects :)

jantje, 2016-05-07:
This one is fixed on my system.
Anyone on the latest nightly still has this problem?

EdwinCroissant, 2016-05-08:
It works again! Thank you :+1: !

jantje, 2016-05-08:
txs for the feedback

closed 2016-08-01 by jantje