Sloeber/arduino-eclipse-plugin #442: Color problem

universpharmacie, 2016-04-25:
Perhaps it's not an issue but silly me, but when selecting the Dark theme in the preferences, most labels in the Project Explorer and in the Outline view are shown black on dark grey. I had a look through all Preference -> Appearance settings which I could find, but I couldn't find the setting to change the foreground color. See attached screen capture
[image]

jantje, 2016-04-25:
I don't think this is related to the plugin. 
I personally don't use the dark theme but I do know others do. 
I also know @dschaefer reported issues in this area on twitter as well https://twitter.com/dougschaefer/status/697868144036200448

jantje, 2016-05-07:
@universpharmacie 
Do you think there is something we can do to help out here?
Or do we close this issue as not relevant to the plugin?

universpharmacie, 2016-05-07:
I had hoped that the eclipse experts here would point me towards information on how to fix color schemes even for parts which can't be configured in the preferences pane (these must be configured too, but elsewhere...). I did not not get the help which I expected here and I am thus deeply disappointed ;-)
But since it is not an issue of the plugin, it may naturally be closed.

jantje, 2016-05-07:
I'm sorry we disappointed you. 
txs we can close this issue

closed 2016-05-07 by jantje

wimjongman, 2016-05-08:
Try to install the plugin spies from [1] and see what CSS is associated with that specific line
Install Jeeeyul's Eclipse Themes from marketplace

[1] http://download.eclipse.org/e4/snapshots/org.eclipse.e4.tools/