Sloeber/arduino-eclipse-plugin #443: V3 64bit Win7: Cannot run program "..../make": Launching failed

tuxedo0801, 2016-05-03:
Downloaded latest V3 64bit Win package. 
Started IDE, waited until all is done,
Created project "testsketch" for Arduino Leonardo with "empty setup/loop sketch".

Clicked on "verify" icon in toolbar.

Result:

```
Error: Cannor run program "O:\eclipseArduino\arduinoPlugin\tools\make\make": Launching failed
```

No wonder... the make executable is missing. The folder "O:\eclipseArduino\arduinoPlugin\tools\make\" is empty.

jantje, 2016-05-03:
Did you install the in O:\eclipseArduino?
Is there anything in O:\eclipseArduino\arduinoPlugin?

tuxedo0801, 2016-05-03:
Yes, extracted the .tar.gz to O:\eclipseArduino

O:\eclipseArduino\arduinoPlugin contains:
[image]

But \tools\make is empty :-(

jantje, 2016-05-03:
Can you check whether there are errors in the log file?
Could it be that a virus scanner removed the file?
As a workaround download make from ftp://ftp.equation.com/make/32/make.exe

tuxedo0801, 2016-05-04:
Attached the log from <workspace>\.metadata\.log -> metadata-log.log.txt
[metadata-log.log.txt](https://github.com/jantje/arduino-eclipse-plugin/files/248349/metadata-log.log.txt)

Virusscanner is running, but had no problem with ArduinoEclipse (checked the logs)

ftp://ftp.equation.com/make/32/make.exe is reachable from my network, and at the time I run ArduinoEclipse, internet-connection was not broken.

jantje, 2016-05-04:
If you look at the log you see there were several network issues for instance

> !SUBENTRY 2 it.baeyens.arduino.core 4 0 2016-05-03 10:50:43.738
> !MESSAGE downloading make.exe
> !STACK 0
> java.net.SocketException: Network is unreachable: connect
> 	at java.net.DualStackPlainSocketImpl.connect0(Native Method)
> 	at java.net.DualStackPlainSocketImpl.socketConnect(Unknown Source)
> 	at java.net.AbstractPlainSocketImpl.doConnect(Unknown Source)
> 	at java.net.AbstractPlainSocketImpl.connectToAddress(Unknown Source)
> 	at java.net.AbstractPlainSocketImpl.connect(Unknown Source)
> 	at java.net.PlainSocketImpl.connect(Unknown Source)
> 	at java.net.Socket.connect(Unknown Source)
> 	at java.net.Socket.connect(Unknown Source)
> 	at sun.net.ftp.impl.FtpClient.openPassiveDataConnection(Unknown Source)
> 	at sun.net.ftp.impl.FtpClient.openDataConnection(Unknown Source)
> 	at sun.net.ftp.impl.FtpClient.getFileStream(Unknown Source)
> 	at sun.net.www.protocol.ftp.FtpURLConnection.getInputStream(Unknown Source)
> 	at java.net.URL.openStream(Unknown Source)
> 	at it.baeyens.arduino.managers.Manager.downloadAndInstall(Manager.java:160)
> 	at it.baeyens.arduino.managers.ArduinoPlatform.install(ArduinoPlatform.java:212)
> 	at it.baeyens.arduino.ui.PlatformPreferencePage.updateInstallation(PlatformPreferencePage.java:127)
> 

Which is here
https://github.com/jantje/arduino-eclipse-plugin/blob/master/it.baeyens.arduino.core/src/it/baeyens/arduino/managers/Manager.java#L160
or downloading the make file.

I find this alarming

> !MESSAGE Failed to extract tar.bz2. O:/eclipseArduino/arduinoPlugin/downloads/esp8266-2.2.0.zip
> !STACK 0
> java.io.IOException: Can't extract file O:\eclipseArduino\arduinoPlugin\packages\esp8266\hardware\esp8266\2.2.0\boards.txt, file already exists!

You can try to fix by downloading make but there may be other problems. If so; stop the plugin remove the "O:\eclipseArduino\arduinoPlugin folder and start the plugin again.

tuxedo0801, 2016-05-04:
Thanks for clarification.

If a download is not working - for whatever reason - I would expect a notification dialog/message. But I'm not aware of any visible error message.

I will start from scratch and check if its working then.

jantje, 2016-05-04:
>If a download is not working - for whatever reason - I would expect a notification dialog/message. 

me to because of this:
https://github.com/jantje/arduino-eclipse-plugin/blob/master/it.baeyens.arduino.core/src/it/baeyens/arduino/managers/Manager.java#L164
But apparently it didn't happen or you missed it.

tuxedo0801, 2016-05-04:
Did a fresh install in Win7 64bit...

make is still missing, but this time without an error in the log. 

[log.txt](https://github.com/jantje/arduino-eclipse-plugin/files/248813/log.txt)

Any ideas?

jantje, 2016-05-04:
Do a fresh install and open a blank workspace. 
look at this video on how to upgrade your workspace
https://www.youtube.com/watch?v=h-rz2FR3H6Y

tuxedo0801, 2016-05-04:
It was indeed a fresh install with a fresh workspace ... 

After working for about 13 yrs with Eclipse, Netbeans, ... I'm very familiar with what those IDEs do and what not. 

This variant simply does not work :-( 

Tried exactly the same on Debian, 64bit. Also does not work. Is hanging forever while installiung/processing file "top_defs.h" from wifishield-lib. No log file in workspace/metadata :-(

jantje, 2016-05-04:
According to the log 
startup at 12:55:09
at 12:57:13 reports a indexing problem of the testproject.
The toolchains can not be downloaded in 2 minutes.
So the log is not from an empty workspace

tuxedo0801, 2016-05-04:
Again:

I removed the complete eclipseArduino folder, which contained also the workspace (this is where eclipse places the default workspace).

Then I extracted the archive (again), started the IDE, waited until nothing visually happens (this has taken some time, not more than about 2mins... my guess), then created the testproject, added a few lines of code (just some serial debug output stuff), clicked in verify... Something happend, no error visible.

Then I switched to the console, and saw that make is again missing.

I have to say that I have a almost 100mbit internet connection. So downloading is normally very fast.

This is the current situation, believe it or not.

jantje, 2016-05-04:
With a 100mbit a 2 minute download can be possible.
In the last log I see no network error reports.
So I see 2 remaining possibilities
1) something removes the make from your system (like virus scanner on your nas)
2) The download fails due to the o drive
Are you using the nightly or the stable?
Can you test on the c drive?
Can you download make and put it at O:\eclipseArduino\arduinoPlugin\tools\make\

tuxedo0801, 2016-05-06:
1) If virusscanner would remove a file, there would be some kind of report, that the file has been removed. But there is none. I even deactivated the virus scanner for the last test. There is no NAS, it's all local hard drive/SSD drive.
2) o Drive is a local SSD drive, where I also run eclipse, netbeans, arduino IDE, ... 

I will retest with c drive. But my guess is, result would be the same :-(

jantje, 2016-05-06:
As windows 64 bit is a very commonly used os with the plugin you are experiencing a exceptional situation if you use the stable.
Please try the workaround I proposed

tuxedo0801, 2016-05-09:
Really strange... Did exactly the same on drive c and now it works?!

Will repeat it (to be sure) on my ssd drive o.

jantje, 2016-05-09:
We have had very strange results with different drives. Even more when cygwin is involved.
Do you have cygwin installed?

tuxedo0801, 2016-05-09:
No cygwin on this machine.

Repeated the step with drive o / ssd. Works... 
Will try another different thing to find the root- cause.

tuxedo0801, 2016-05-09:
I think I found the root-cause:

There's an Kaspersky Antivirus on this machine. If virus scanner is active, it's not working. The vorus scanner does not raise an alarm or notification, and also there's no log entry that something has been blocked (otherwise our admin would have contacted me).

If I disable the virus scanner, it works. 

---> Very strange.

Have to repeat the test on debian linux (there I also had problems with the same workflow, but they are slightly different)

jantje, 2016-05-09:
Txs for the feedback
So for windows we are back to virus scanner.

tuxedo0801, 2016-05-09:
Which is still strange because:

* there's no warning that something happened on side of virus scanner
* I'm sure that I have tested few days ago without virus scanner, which also didn' work

jantje, 2016-05-09:
>I'm sure that I have tested few days ago without virus scanner, which also didn' work

I've been there. Later I go: Ooooch yeah, what possibly happened was ......
That is mostly the moment I start to realize I made a mistake and I learned another lesson.

rlogiacco, 2016-05-10:
I've had a similar issue with virus scanners, apparently detecting `make.exe` as a critical software to protect against :smiling_imp: 

We have gone through extended testing on Windows 64bit with multiple flavours of Windows and we are pretty confident the Arduino Eclipse Plugin works nicely... all the troubles we have identified so fare were coming from other softwares, including (as an example) a Git installation.

It's very good to know that you were so interested in the plugin you actually spent quite some time trying to identify the root cause :wink:

jantje, 2016-05-17:
@tuxedo0801 
Can we close this as a virus scanner related issue?

jantje, 2016-05-29:
I'm closing this as a virus scanner related issue

closed 2016-05-29 by jantje