Sloeber/arduino-eclipse-plugin #446: How to restore Serial Monitor's CLEAR button?

adamryczkowski, 2016-05-11:
When I run ArduinoEclipse stand-alone product, the Serial Monitor is lacking much of its functionality. In particular there is no way to clear its contents: 
[image]

For reference, this is what I can see on the Serial Monitor, when I installed Arduino IDE V3 over already existing Eclipse installation:
[image]

adamryczkowski, 2016-05-11:
OK. I found the solution myself, sorry for taking your time.

The answer to those missing buttons is in those small toolbar icons on the top of SerialMonitor window, just beside the "+" and "x" buttons...

closed 2016-05-11 by adamryczkowski

jantje, 2016-05-11:
good work :-)

adamryczkowski, 2016-05-12:
No. It is your work that is truly awesome. Thank you!