Sloeber/arduino-eclipse-plugin #447: Libraries location keeps resetting

OneManOneClass, 2016-05-12:
I'm using some custom libraries for my Arduino projects and I've added the paths in Preferences/Arduino and add them to my projects the usual (probably) way by going to the Arduino tab and selecting "Add a library to the selected project". Everything goes fine until I restart the Eclipse. Then, when I try to add libraries, it only displays the default ones. If I want the IDE to start showing my libraries again, I have to go back to Prefferences/Arduino and close the window by pressing Ok. Only then it starts displaying my libraries again.

jantje, 2016-05-12:
That is weird
[image]
.
I have multiple private libraries and I never experienced this problem.

Unfortunately I can not help you as I decided to no longer respond to github issues -and other questions- due to a lack of support.
As a patron you can post a message on the patreon forum stating you support a github issue. This allows me to support people who support me (starting from as low as 1$ a month exclusive VAT).
https://www.patreon.com/posts/3700469

universpharmacie, 2016-05-12:
I've the same problem. Since there is the simple workaround to go into preferences to make the IDE enumerating and generating the libraries menu again, I first thought that it was not worth mentioning it, but since I'm not alone...

jantje, 2016-05-12:
@universpharmacie 
If I'm not mistaken you are a patron of mine.

universpharmacie, 2016-05-12:
Yes I am, but I can wait until OneManOneClass becomes a patron, too. No hurry! ;)

OneManOneClass, 2016-05-12:
Sadly, I'm a college student and don't have a loose penny :(

universpharmacie, 2016-05-12:
In that case and since I have the same issue, and if it will be ok for Jantje, I'm ready to take this on my bierviltje... ;)

jantje, 2016-05-12:
I'm just wondering. Could this be a duplicate of #409?
409 has been fixed on the 3th of april so is not part of the stable release.
409 is for sure a problem on mac but IMHO also on linux

universpharmacie, 2016-05-12:
I'm using the nightly build. I'll check out again (haven't used that functionality in the last weeks) and write back.

universpharmacie, 2016-05-12:
ok, I stand corrected. Seems to fixed in the nightly build.

OneManOneClass, 2016-05-12:
My apologies, before posting the issue I did try to search but found nothing. Thank you for your responses, I'll give the nightly build a try as soon as I'm done with my current project! (Will have to be going to preferences until then...)

jantje, 2016-05-12:
@OneManOneClass are you on mac or linux?
I'll mark it as a duplicate of #409 for now

OneManOneClass, 2016-05-12:
I'm actually on Windows 7 64-bit.

jantje, 2016-05-12:
That is wierd as I am to and I do not have the issue.
Can you share the folder names?

OneManOneClass, 2016-05-12:
[image]

jantje, 2016-05-12:
You have a mix of forward and backward slashes. That is at least weird.

OneManOneClass, 2016-05-12:
The directory located in C: was there by default. I've reattached it and now it's also with backward slashes instead of forward ones.  Problem persists.

jantje, 2016-05-12:
I have no access to my windows machine until next week. I see however nothing special (except for the slashes) that could explain what is wrong.

OneManOneClass, 2016-05-12:
Well, as I've wrote before, once I finish with my current project, I'll give the nightly build a shot and see if the problem is still there. Thought about slashes being a problem (yet the location is selected, not typed in) but figured that if the IDE simply is unable to find paths, going to preferences shouldn't fix it.

jantje, 2016-05-12:
>Thought about slashes being a problem (yet the location is selected, not typed in) but figured that if the IDE simply is unable to find paths, going to preferences shouldn't fix it. 

you are making assumptions. The plugin does not validate the content. It has to wrap and unwrap the list for storage. #409 fixes a bug about inconsistent wrapping versus unwrapping. However windows worked fine so I do not think the nightly will help you.

mdilger, 2016-05-13:
I am also running on Windows 7 but have not noticed this problem, maybe due to the fact that I am not using private libraries on my current project, but I am not sure .  I will take a closer look tonight and let you know my findings.

mdilger, 2016-05-13:
Looks **OK** on my Windows 7 64 bit computer.

My private library path is: 

C:\Development\Libraries

jantje, 2016-05-13:
@mdilger thanks for the input

jantje, 2016-05-17:
@OneManOneClass 
I tried this here on my windows machine. Works fine. Is there anything special in your setup?

jantje, 2016-05-29:
as @OneManOneClass no longer responds I'm closing this issue as a usage error

closed 2016-05-29 by jantje