Sloeber/arduino-eclipse-plugin #448: platform.txt for PJRC Teensy

universpharmacie, 2016-05-12:
After updating the Teensyduino Add-on, it is still needed to edit platform.txt manually to get everything working with the Eclipse Arduino plugin.

Instead of `compiler.path={runtime.ide.path}/`
I have to put `compiler.path={runtime.hardware.path}/../../`

and instead of `tools.teensyloader.cmd.path={runtime.ide.path}/hardware/tools/teensy_post_compile`
I have to put `tools.teensyloader.cmd.path={runtime.hardware.path}/../tools/teensy_post_compile`

There must be something wrong with the {runtime.ide.path}, at least in OSX, but I don't know if it's Paul's or Jantje's fault.

jantje, 2016-05-12:
In windows this is fixed in 1.2.8.
See the discussion https://forum.pjrc.com/threads/33405-Teensyduino-1-28-Beta-1-Available?p=101256&viewfull=1#post101256

universpharmacie, 2016-05-12:
All I can see is that it's not ok for OSX in 1.2.9 beta 1 (and it wasn't ok in 1.2.8)

universpharmacie, 2016-05-12:
I went back to the original platform.txt file from Teensy 1.2.9 and from the error messages, I find that {runtime.ide.path} seems to be empty or invisible for the eclipse Arduino plugin since I get: 
Program "/hardware/tools/arm/bin/arm-none-eabi-g++" not found in PATH.

So I found an easier workaround: Add the following line at the beginning of platform.txt
`runtime.ide.path="/Applications/arduino.app/Contents/Java"`
(which works naturally only on Mac) and which will fix all consecutive relative paths with are based on {runtime.ide.path} .

But it's still not funny and I'd like to get a real fix.

universpharmacie, 2016-05-12:
mentionner the issue on [https://forum.pjrc.com/threads/34472-Teensyduino-1-29-Beta-1-Available?p=104260&viewfull=1#post104260] , too.

jantje, 2016-05-12:
As you know I don't know mutch about mac. So I need to ask.
Is `runtime.ide.path="/Applications/arduino.app/Contents/Java"` the only option on mac? Or is this the default option?

universpharmacie, 2016-05-12:
This is the default installation path of the Arduino IDE on Mac. Despite the eclipse Arduino plugin having no more Arduino dependencies, an Arduino install is still needed to host the Teensyduino extension.
I think that 99% of Mac users will have it in the default install location, theoretically, it could be `/whatEverPath/arduino.app/Contents/Java` or still worse `/Volumes/whatEverVolume/whatEverPath/arduino.app/Contents/Java`.

jantje, 2016-05-12:
>This is the default installation path of the Arduino IDE on Mac. 

That is what I thought.

>Despite the eclipse Arduino plugin having no more Arduino dependencies, an Arduino install is still needed to host the Teensyduino extension.

I know. It is a decision of paul to not support the arduino boards manager. 
I just installed teensyduino 1.2.9 beta (on windows) and I see
```
compiler.path={runtime.hardware.path}/../tools/
tools.teensyloader.cmd.path={runtime.hardware.path}/../tools
```
I think it is really strange there is a different platform file for mac and windows

jantje, 2016-05-12:
Did you also notice that you have changed to another path? ../tools versus ../..

universpharmacie, 2016-05-12:
ok, I'll trash everything and restart with a clean Arduino IDE and Teensyduino install. I think that several things may have been messed up over the time. Will write back later.

universpharmacie, 2016-05-12:
After a fresh install, the problem does not longer persist. I guess that when installing Teensyduino 1.2.9 over my old stuff, the platform.txt file was not overwritten (for whatever reason). Now everything works like a charm without patching. Please accept my apologies for having waisted your precious time!

jantje, 2016-05-12:
>from where does that (in)famous runtime.ide.path variable come?

Before the boardsmanager this was created by the arduino IDE to refer to where the arduino IDE was installed. adding hardware/[platform] lead to the .txt files.
The arduino plugin made this variable based on which arduino ide was referenced.

When the boardsmanager was introduced the arduino ide team decided to go to 2 locations to store hardware. This made it impossible to keep referencing runtime.ide.path as the root of all hardware.
So they dropped runtime.ide.path

There are and always have been plenty of variables that are referenced but not defined in the .txt files. 
It is one of the more obscure parts of the arduino ide.

>Please accept my apologies for having waisted your precious time!

accepted :-)

closed 2016-05-12 by jantje