Sloeber/arduino-eclipse-plugin #452: The V3 Product not usable under Ubuntu 16.04

adamryczkowski, 2016-05-13:
After a fresh install of the product, after adding at least one Arduino project, it keeps Eclipse consuming CPU 100% and it prevents basically any action, except closing the program. In particular, eclipse update and preferences window hang indefinitely, as if they were waiting for some background task to finish (which might be the reason, why CPU is 100% busy). 

I tried OpenJDK 8 and Oracle Java 8. I reinstalled the project from scratch under different users' profiles. 

I wish I could tell you which particular task is taking the CPU, but unfortunately all eclipse's tasks under `htop` are displayed exactly the same. 

How to reproduce:

1. Install Ubuntu 16.04 64 bit,
2. Install java
3. Download the product and install it on any user-writable directory
4. Run and add a arduino project
5. Try changing something in the preferences or upgrade the eclipse

I am a professional programmer, but I know nothing about development of Java applications. If you show me what to do to, I'll help you further diagnose the problem.

adamryczkowski, 2016-05-13:
I tried plain Eclipse on the same machine.

The good news is that on Ubuntu 16.04 the plugin works correctly (c++ indexing works), so as a walkaround you can just use Eclipse's installer, install eclipse for C++ developers, and then, on Eclipse, install the Arduino V3 plugin.

adamryczkowski, 2016-05-13:
When I use vanilla Eclipse with CDT and without the Arduino IDE, I still see almost 100% cpu utilization, but the platform itself is responsive. I can change global preferences and check for updates.

adamryczkowski, 2016-05-13:
It might be related to the [https://www.eclipse.org/forums/index.php/t/1076269/](url)

adamryczkowski, 2016-05-13:
The bug is caused by incompatibility of old Eclipse with new QT shipped by Ubuntu.

Upgrading to Eclipse Neon did solves the problem. Jantie, you should mention it for other Ubuntu 16.04 users.

closed 2016-05-13 by adamryczkowski

jantje, 2016-05-13:
Thanks I'llmod the web