Sloeber/arduino-eclipse-plugin #453: How to import project?How to use Eclipse Arduino IDE compiler Marlin?!

BCCW, 2016-05-16:


BCCW, 2016-05-16:
https://github.com/MarlinFirmware/Marlin

jantje, 2016-05-16:
What is the problem?

BCCW, 2016-05-17:
Hi jantje，Thank you for your reply!!!
If you use the Arduino IDE to open the Marlin project files "marlin.ino", all other relevant documents will be automatically opened, and you will not have problems compiling! But if you use Eclipse Arduino IDE open the Marlin project files "marlin.ino",it will only open the "marlin.ino", can not compile all project files! If you use Eclipse Arduino IDE to create a "New Arduino sketch",and"Import Marlin folder containing source code in the current project" ,it has an error when you compile it.(error:multiple definition of 
 `__vector_17'),So I ask you：How to import project? How to use Eclipse Arduino IDE compiler Marlin?! Thank you!

jantje, 2016-05-17:
I decided to no longer respond to github issues -and other questions- due to a lack of support.
As a patron you can post a message on the patreon forum stating you support a github issue. This allows me to support people who support me (starting from as low as 1$ a month exclusive VAT).
https://www.patreon.com/posts/3700469

closed 2016-08-20 by jantje