Sloeber/arduino-eclipse-plugin #458: V3; W7

FermiSCM, 2016-05-21:
Hello, I have following issues, can someone help me, please - Thanks foward
Description	Resource	Path	Location	Type
make: *** [.ino.cpp.o] Error 2
Program "D:\downloadPrg\eclipseArduino\arduinoPlugin\tools\arduino\arm-none-eabi-gcc\4.8.3-2014q1\bin\arm-none-eabi-g++" not found in PATH	test01		Project Properties, C++ Preprocessor Include.../Providers, [Arduino Compiler Settings] options	C/C++ Scanner Discovery Problem

Program "D:\downloadPrg\eclipseArduino\arduinoPlugin\tools\arduino\arm-none-eabi-gcc\4.8.3-2014q1\bin\arm-none-eabi-gcc" not found in PATH	test01		Project Properties, C++ Preprocessor Include.../Providers, [Arduino Compiler Settings] options	C/C++ Scanner Discovery Problem

recipe for target '.ino.cpp.o' failed	subdir.mk	/test01/Release	line 24	C/C++ Problem

jantje, 2016-05-21:
I decided to no longer respond to github issues -and other questions- due to a lack of support.
As a patron you can post a message on the patreon forum stating you support a github issue. This allows me to support people who support me (starting from as low as 1$ a month exclusive VAT).
https://www.patreon.com/posts/3700469

Apart from that . The information you provided is not enough to investigate.
Did you use the nightly?
Is this a newly created project?
Are there errors in the error log?
Can you provide the full console output
Have you waited long enough to create the project after the installation of the board?
.....

FermiSCM, 2016-05-21:
Hi Jantje, fine I will support you on patreon, I need help.Thanks

jantje, 2016-05-21:
cool thanks :-)
As far as I can see from your post you are trying to build for due or zero (sam or samd)
Both of them need to be installed after the initial setup. After the preference page is closed the boards are not yet installed. On my system in the bottom right you can see the progress of the install.
You can also see this if you do window->view->other->progress.
As long as the installation is ongoing you should not create sketches/projects.
If you did so go into the project properties->arduino and select apply.
try again.

FermiSCM, 2016-05-21:
Yes I am using arduino due; I followed your video; I installed in properties->arduino->boardt.txt file: 1.6.8\boards.txt and board arduino due; but still have:  make: *** [.ino.cpp.o] Error 1  test_ms   C/C++ Problem

jantje, 2016-05-21:
can you provide the whole output in the console?

FermiSCM, 2016-05-21:
In the console is nothing in.

jantje, 2016-05-21:
[image]
The console windows is a collection of consoles. As you have errors the "CDT Build Console [project name]" must contain data. You can switch console by selecting the down arrow right of the blue monitor on the right top of the console view
mine contains this
```
3:27:29 **** Build of configuration Release for project speedtest ****
"C:\\embedded\\V3.0\\arduinoPlugin\\tools\\make\\make" -j4 all 
'Building file: ../.ino.cpp'
'Building file: F:/windows/arduino/teensy 1.28 beta/hardware/teensy/avr/cores/teensy3/AudioStream.cpp'
'Building file: F:/windows/arduino/teensy 1.28 beta/hardware/teensy/avr/cores/teensy3/DMAChannel.cpp'
'Starting C++ compile'
'Starting C++ compile'
"F:/windows/arduino/teensy 1.28 beta/hardware/teensy/../tools/arm/bin/arm-none-eabi-g++" -c -Os -g -Wall -ffunction-sections -fdata-sections -nostdlib -MMD -fno-exceptions -felide-constructors -std=gnu++0x -fno-rtti -mthumb -mcpu=cortex-m4 -fsingle-precision-constant -D__MK20DX256__ -DTEENSYDUINO=128 -DARDUINO=10606 -DF_CPU=72000000 -DARDUINO_ARCH_AVR -DUSB_SERIAL -DLAYOUT_GERMAN_MAC   -I"F:\windows\arduino\teensy 1.28 beta\hardware\teensy\avr\cores\teensy3" -MMD -MP -MF".ino.cpp.d" -MT".ino.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "../.ino.cpp"  -o  ".ino.cpp.o"   -Wall
'Starting C++ compile'
'Building file: F:/windows/arduino/teensy 1.28 beta/hardware/teensy/avr/cores/teensy3/HardwareSerial1.cpp'
"F:/windows/arduino/teensy 1.28 beta/hardware/teensy/../tools/arm/bin/arm-none-eabi-g++" -c -Os -g -Wall -ffunction-sections -fdata-sections -nostdlib -MMD -fno-exceptions -felide-constructors -std=gnu++0x -fno-rtti -mthumb -mcpu=cortex-m4 -fsingle-precision-constant -D__MK20DX256__ -DTEENSYDUINO=128 -DARDUINO=10606 -DF_CPU=72000000 -DARDUINO_ARCH_AVR -DUSB_SERIAL -DLAYOUT_GERMAN_MAC   -I"F:\windows\arduino\teensy 1.28 beta\hardware\teensy\avr\cores\teensy3" -MMD -MP -MF"core/AudioStream.cpp.d" -MT"core/AudioStream.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "F:/windows/arduino/teensy 1.28 beta/hardware/teensy/avr/cores/teensy3/AudioStream.cpp"  -o  "core/AudioStream.cpp.o"   -Wall
"F:/windows/arduino/teensy 1.28 beta/hardware/teensy/../tools/arm/bin/arm-none-eabi-g++" -c -Os -g -Wall -ffunction-sections -fdata-sections -nostdlib -MMD -fno-exceptions -felide-constructors -std=gnu++0x -fno-rtti -mthumb -mcpu=cortex-m4 -fsingle-precision-constant -D__MK20DX256__ -DTEENSYDUINO=128 -DARDUINO=10606 -DF_CPU=72000000 -DARDUINO_ARCH_AVR -DUSB_SERIAL -DLAYOUT_GERMAN_MAC   -I"F:\windows\arduino\teensy 1.28 beta\hardware\teensy\avr\cores\teensy3" -MMD -MP -MF"core/DMAChannel.cpp.d" -MT"core/DMAChannel.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "F:/windows/arduino/teensy 1.28 beta/hardware/teensy/avr/cores/teensy3/DMAChannel.cpp"  -o  "core/DMAChannel.cpp.o"   -Wall
'Starting C++ compile'
"F:/windows/arduino/teensy 1.28 beta/hardware/teensy/../tools/arm/bin/arm-none-eabi-g++" -c -Os -g -Wall -ffunction-sections -fdata-sections -nostdlib -MMD -fno-exceptions -felide-constructors -std=gnu++0x -fno-rtti -mthumb -mcpu=cortex-m4 -fsingle-precision-constant -D__MK20DX256__ -DTEENSYDUINO=128 -DARDUINO=10606 -DF_CPU=72000000 -DARDUINO_ARCH_AVR -DUSB_SERIAL -DLAYOUT_GERMAN_MAC   -I"F:\windows\arduino\teensy 1.28 beta\hardware\teensy\avr\cores\teensy3" -MMD -MP -MF"core/HardwareSerial1.cpp.d" -MT"core/HardwareSerial1.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "F:/windows/arduino/teensy 1.28 beta/hardware/teensy/avr/cores/teensy3/HardwareSerial1.cpp"  -o  "core/HardwareSerial1.cpp.o"   -Wall
'Finished building: F:/windows/arduino/teensy 1.28 beta/hardware/teensy/avr/cores/teensy3/AudioStream.cpp'
'Finished building: ../.ino.cpp'
'Finished building: F:/windows/arduino/teensy 1.28 beta/hardware/teensy/avr/cores/teensy3/DMAChannel.cpp'
'Finished building: F:/windows/arduino/teensy 1.28 beta/hardware/teensy/avr/cores/teensy3/HardwareSerial1.cpp'
' '
' '
' '
' ' 
```

FermiSCM, 2016-05-21:
I installed complet new and I followed you advise, now it works but I am wondering because I have errors
I use the example Blink:
Symbol 'OUTPUT' could not be resolved	Blink.ino	/test1	line 23	Semantic Error
Symbol 'LOW' could not be resolved	Blink.ino	/test1	line 31	Semantic Error
Symbol 'HIGH' could not be resolved	Blink.ino	/test1	line 29	Semantic Error
Invalid arguments ' Candidates are: void pinMode(long int, long int) 	Blink.ino	/test1	line 23	Semantic Error
Invalid arguments ' Candidates are: void digitalWrite(long int, long int) 	Blink.ino	/test1	line 29	Semantic Error
Invalid arguments ' Candidates are: void digitalWrite(long int, long int) 	Blink.ino	/test1	line 31	Semantic Error
'begin' is ambiguous ' Candidates are: void begin(unsigned long int) void begin(long int)' Blink.ino	/test1	line 20	Semantic Error
I add only in the code: Serial.begin(9600);
Serial.print ("TEST_01");

The console shows:
'Finished building: test01.bin'
' '
'Building target: test01'
'Printing size:'
"D:\downloadPrg\eclipseArduino\arduinoPlugin\tools\arduino\arm-none-eabi-gcc\4.8.3-2014q1/bin/arm-none-eabi-size" -A "D:/downloadPrg/eclipseArduino/workspace/test01/Release/test01.elf"
D:/downloadPrg/eclipseArduino/workspace/test01/Release/test01.elf  :
section             size        addr
.text              22204      524288
.ARM.exidx             8      546492
.relocate           2260   537329664
.bss                2600   537331924
.ARM.attributes       41           0
.comment             112           0
.debug_info        74163           0
.debug_abbrev      13398           0
.debug_aranges      1728           0
.debug_ranges       2352           0
.debug_line        23595           0
.debug_str         19758           0
.debug_frame        6628           0
.debug_loc         15235           0
Total             184082

'Finished building target: test01'
' '

18:04:25 Build Finished (took 1s.663ms)

18:06:21 **** Clean-only build of configuration Release for project test02 ****
"D:\\downloadPrg\\eclipseArduino\\arduinoPlugin\\tools\\make\\make" clean 
make: *** No rule to make target 'clean'.  Stop.

18:06:21 Build Finished (took 150ms)

18:06:22 **** Build of configuration Release for project test02 ****
"D:\\downloadPrg\\eclipseArduino\\arduinoPlugin\\tools\\make\\make" all 
'Building file: ../.ino.cpp'
'Starting C++ compile'
"D:\downloadPrg\eclipseArduino\arduinoPlugin\tools\arduino\arm-none-eabi-gcc\4.8.3-2014q1/bin/arm-none-eabi-g++" -c -g -Os -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -Dprintf=iprintf -MMD -mcpu=cortex-m3 -mthumb -DF_CPU=84000000L -DARDUINO=10606 -DARDUINO_SAM_DUE -DARDUINO_ARCH_SAM  -D__SAM3X8E__ -mthumb -DUSB_VID=0x2341 -DUSB_PID=0x003e -DUSBCON  -DUSB_MANUFACTURER=\""Arduino LLC"\" -DUSB_PRODUCT=\""Arduino Due"\" "-ID:/downloadPrg/eclipseArduino/arduinoPlugin/packages/arduino/hardware/sam/1.6.8/system/libsam" "-ID:/downloadPrg/eclipseArduino/arduinoPlugin/packages/arduino/hardware/sam/1.6.8/system/CMSIS/CMSIS/Include/" "-ID:/downloadPrg/eclipseArduino/arduinoPlugin/packages/arduino/hardware/sam/1.6.8/system/CMSIS/Device/ATMEL/"   -I"D:\downloadPrg\eclipseArduino\arduinoPlugin\packages\arduino\hardware\sam\1.6.8\cores\arduino" -I"D:\downloadPrg\eclipseArduino\arduinoPlugin\packages\arduino\hardware\sam\1.6.8\variants\arduino_due_x" -MMD -MP -MF".ino.cpp.d" -MT".ino.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "../.ino.cpp"  -o  ".ino.cpp.o"   -Wall
'Finished building: ../.ino.cpp'
' '
'Building file: D:/downloadPrg/eclipseArduino/arduinoPlugin/packages/arduino/hardware/sam/1.6.8/variants/arduino_due_x/variant.cpp'
'Starting C++ compile'
"D:\downloadPrg\eclipseArduino\arduinoPlugin\tools\arduino\arm-none-eabi-gcc\4.8.3-2014q1/bin/arm-none-eabi-g++" -c -g -Os -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -Dprintf=iprintf -MMD -mcpu=cortex-m3 -mthumb -DF_CPU=84000000L -DARDUINO=10606 -DARDUINO_SAM_DUE -DARDUINO_ARCH_SAM  -D__SAM3X8E__ -mthumb -DUSB_VID=0x2341 -DUSB_PID=0x003e -DUSBCON  -DUSB_MANUFACTURER=\""Arduino LLC"\" -DUSB_PRODUCT=\""Arduino Due"\" "-ID:/downloadPrg/eclipseArduino/arduinoPlugin/packages/arduino/hardware/sam/1.6.8/system/libsam" "-ID:/downloadPrg/eclipseArduino/arduinoPlugin/packages/arduino/hardware/sam/1.6.8/system/CMSIS/CMSIS/Include/" "-ID:/downloadPrg/eclipseArduino/arduinoPlugin/packages/arduino/hardware/sam/1.6.8/system/CMSIS/Device/ATMEL/"   -I"D:\downloadPrg\eclipseArduino\arduinoPlugin\packages\arduino\hardware\sam\1.6.8\cores\arduino" -I"D:\downloadPrg\eclipseArduino\arduinoPlugin\packages\arduino\hardware\sam\1.6.8\variants\arduino_due_x" -MMD -MP -MF"core/variant.cpp.d" -MT"core/variant.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "D:/downloadPrg/eclipseArduino/arduinoPlugin/packages/arduino/hardware/sam/1.6.8/variants/arduino_due_x/variant.cpp"  -o  "core/variant.cpp.o"   -Wall
'Finished building: D:/downloadPrg/eclipseArduino/arduinoPlugin/packages/arduino/hardware/sam/1.6.8/variants/arduino_due_x/variant.cpp'
' '
'Building file: D:/downloadPrg/eclipseArduino/arduinoPlugin/packages/arduino/hardware/sam/1.6.8/cores/arduino/avr/dtostrf.c'
'Starting C compile'
"D:\downloadPrg\eclipseArduino\arduinoPlugin\tools\arduino\arm-none-eabi-gcc\4.8.3-2014q1/bin/arm-none-eabi-gcc" -c -g -Os -std=gnu11 -ffunction-sections -fdata-sections -nostdlib --param max-inline-insns-single=500 -Dprintf=iprint [... cut ...]
.bss                2600   537331924
.ARM.attributes       41           0
.comment             112           0
.debug_info        74159           0
.debug_abbrev      13413           0
.debug_aranges      1728           0
.debug_ranges       2352           0
.debug_line        23595           0
.debug_str         19757           0
.debug_frame        6628           0
.debug_loc         15235           0
Total             184084

'Finished building target: test1'
' '

18:15:22 Build Finished (took 1s.690ms)

FermiSCM, 2016-05-21:
I uploaded to the due and it works with errors?
project-> properties -> arduino -> boardt 1.6.8 and board due

jantje, 2016-05-21:
I wrote a post explaining in depth what this non-issue is all about https://www.patreon.com/posts/there-are-bugs-5507246

FermiSCM, 2016-05-28:
Hi Jantje, thanks for your help. I reached a level that it works. I do not know if I have to open  a new issue. 
Because I have some question to use program code. I had a program working on Atmel studio 6.2. like:

the eclipse V3 don´t resolved following: 
	/**/associate clock with Port A,B,C so we can read input
	PMC->PMC_PCER0 = _BV(ID_PIOA);  // #define _BV(bit) (0x1u << bit)
	PMC->PMC_PCER0 = _BV(ID_PIOB);
	PMC->PMC_PCER0 = _BV(ID_PIOC);
	PMC->PMC_PCER0 = _BV(ID_PIOD);**
even that not
	//Enable PIN !
	PIOA->PIO_PER = PIO_PA15;
it is not possible to use these to read or write register?

jantje, 2016-05-28:
@FermiSCM 
Amtel studio is a completely different beast to Arduino IDE. They also use different libraries.

jantje, 2016-06-22:
@FermiSCM 
Are you ok that I close this one?

FermiSCM, 2016-06-22:
yes thank you again, I hope someday we can meet togher.

Michele

jantje, 2016-06-22:
thanks for the feedback

closed 2016-06-22 by jantje