Sloeber/arduino-eclipse-plugin #460: Upload tool flags discordant with board upload settings for ESP8266

overtone1000, 2016-05-24:
This error was encountered with:
Windows 10, 64-bit
Eclipse Mars 2 (4.5.2RC3)
Arduino Eclipse Extensions v.3.0.0.201603312133

When attempting to upload a sketch to a Sparkfun Thing Dev using the following settings in the board manager:
Board: Generic ESP8266 Moduel
Flash mode: DIO
Flash freq: 80 MHz
CPU freq: 80 MHz
Flash size: 512K (64K SPIFFS)
Debug port: Diabled
Debug level: None
**Reset method: nodemcu**
Upload speed: 921600

The Eclipse console output for an attempted upload contains the following call to the upload tool:

`C:\Program Files\eclipse mars 2rc3\arduinoPlugin\tools\esp8266\esptool\0.4.8/esptool.exe -cd ck -cb 921600 -cp COM3 -ca 0x00000 -cf E:\Dropbox\My Programs\Git Repositories\TRM_Eclipse\TRM_ESP8266_Basic/ESP_8266/TRM_ESP8266_Basic.bin `

However, the **-cd ck** flag does not reflect my selection of the **nodemcu** reset method. Indeed, when I change the flag and run the following on the command line, the upload succeeds:

`"C:\Program Files\eclipse mars 2rc3\arduinoPlugin\tools\esp8266\esptool\0.4.8/esptool.exe" -cd nodemcu -cb 921600 -cp COM3 -ca 0x00000 -cf "E:\Dropbox\My Programs\Git Repositories\TRM_Eclipse\TRM_ESP8266_Basic/ESP_8266/TRM_ESP8266_Basic.bin"`

Other pertinent details:
-The Arduino IDE upload to the board also succeeds with the settings listed above, and correctly places the **-cd nodemcu** flag.
-I'm using the boards.txt file the plugin acquired from _http://arduino.esp8266.com/stable/package_esp8266com_index.json_ and stored in my eclipse directory at _\arduinoPlugin\packages\esp8266\hardware\esp8266\2.2.0\boards.txt_. This version of the file is identical to the one the Arduino IDE acquired.
-I did change the plugin settings to build with the MinGW version of make as the bundled version was giving me compile errors. The plugin does compile the sketch successfully, and the program worked when I directly uploaded the resulting bin file.

jantje, 2016-05-24:
@overtone1000 
Thanks for your support :-)
I tried it here on my system (windows 10  64bit) and I did get the -cd nodemcu
The json file I use is the same as yours and I also use V2.2 I am on the latest version.
The problem is with the guy who makes these boards.txt files.
The upload sqettings are defined 2 times

> generic.menu.ResetMethod.ck=ck
generic.menu.ResetMethod.ck.upload.resetmethod=ck
generic.menu.ResetMethod.nodemcu=nodemcu
generic.menu.ResetMethod.nodemcu.upload.resetmethod=nodemcu

and 
>generic.upload.tool=esptool
generic.upload.speed=115200
generic.upload.resetmethod=ck
generic.upload.maximum_size=434160
generic.upload.maximum_data_size=81920
generic.upload.wait_for_upload_port=true

So this is a duplicate of #439 Which is fixed in the nightly.

You can upgrade to the nightly or you can delete 
>generic.upload.resetmethod=ck
in the boards.txt file and open your project properties and select apply.
Best regards
Jantje

overtone1000, 2016-05-24:
Thank you, Jantje. Sorry for the duplicate, I should have recognized #439 as the same issue. I will switch to the nightly builds.

jantje, 2016-05-25:
your welcome.
You have spend time in documenting the issue. I didn't recognize your issue immediately as a duplicate of #439.  It was only when I saw it worked at my place and I started thinking what I had changed I realized what could be the issue.
Anyway If everybody documented their issues as you did here (and was a patron) life would be far better for me :-)

jantje, 2016-07-05:
Closing this item as it is a duplicate

closed 2016-07-05 by jantje