Sloeber/arduino-eclipse-plugin #461: Library search more restrictive with Eclipse plugin than with Arduino IDE

jrobert-gh, 2016-05-25:
This prevents compiling code that compiles properly under the Arduino IDE. The Blynk library is one example; the Eclipse plugin won't compile Blynk programs because it doesn't find all of Blynk's .h files. Could this restriction be relaxed to be compatible with the Arduino IDE?

jantje, 2016-05-25:
Hi Robert.
There is a huge difference between how arduino IDE find its libraries and how the plugin finds it's libraries. I think the term "more restrictive" does not apply here. It is different and it is different for a good reason.

Personally I find that the way Arduino IDE finds the libraries is opening the door for all kind of hard to troubleshoot problems.  I experience this myself, after Arduino ignored my remarks they experienced it themselves. At that point they started to improve the algorithm.
Arduino ide does it as follows:
If a include is not found in the current project: go and find a folder in the libraries folder **that contains a file with that name**. Add that folder to the include path.

The way the plugin does it:
If a include is not found in the current project: remove the .h part from the include file name and look for a **folder with that name** in the libraries folders. Add that folder to the include path.

There used to be a time that having a library called test containing Arduino.h would make that you simply couldn't build any sketches.
Arduino IDE has improved a lot at this level but the problem is that the foundation is bad. The better you make the code the harder the issue will be to fix.
I just tried here to compile a servo example after I created a folder called test containing a file Servo.h
Aduino IDE 1.6.9 reports (I had to look carefully to see it)

> Multiple libraries were found for "Servo.h"
 Used: F:\windows\arduino\arduino-1.6.9\libraries\Servo
 Not used: C:\Users\jan\Documents\Arduino\libraries\test

Arduino IDE 1.6.5 R2 reports nothing.

In other words the Arduino IDE way of looking up libraries is something I do not want to support. Well at least not with the current support of the community for me.

jrobert-gh, 2016-05-25:
I see the issue you're describing. So I wonder if there isn't some way
to #include a specific library that doesn't allow a world-wide search
for a matching name, but lets you point right at it? My ultimate goal is
Blynk (in this case) sketches that will compile in either environment.
Between the developer and the programmer, is that doable?

JRobert

jantje, 2016-05-25:
The good solution is:
If the folder name is Blynk create a file called Blynk.h in the folder and add
#include "Blynk.h" 
in the example

you can also do menu->arduino->include library->select Blynk

jrobert-gh, 2016-05-25:
Thanks, Jantje. I'll try it the second way; the library folder, Blynk,
has many .h files in it.

JRobert

jantje, 2016-05-26:
> the library folder, Blynk, has many .h files in it.

This is not important for the first solution. For the first solution the question is: "Can I convince the author of the library to add a (empty is ok) Blynk.h file? And more importantly: Is the Author willing to add #include "Blynk.h" to all the examples.
This has no negative impact on the hex file, The working of Arduino IDE (should be more reliable) and it makes the plugin capable of doing auto Include of the library.
It is less than 5 minutes work, but it is the author of the lib his call.

jantje, 2016-05-26:
I had another look at the Blynk github repository and it is not clear to me whether it needs to be Blynk.h or Blynk-library.h

jrobert-gh, 2016-05-26:
Thank you Jantje, you've made me one happy camper!
I can fix this in my own copy of new Blynk library versions in a few
seconds, and the ability to develop with Eclipse is a huge gain for me!

I'd like to pass this on to other users, and to request the Blynk
developers to adopt it. May I copy (with credit to you, of course) your
message into a post on the Blynk forum for this purpose?

JRobert

jantje, 2016-05-26:
Sure you can

jrobert-gh, 2016-08-22:
As this is a "known limitation" it's fine with me to close the issue.

jantje, 2016-08-22:
txs

closed 2016-08-22 by jantje