Sloeber/arduino-eclipse-plugin #465: missing rm command on windows makes clean to fail.

jantje, 2016-05-30:
On windows when you do project clean the  project does not get cleaned because the rm command does not exist.
As a workaround one can delete the folder in the project with the configuration name
I see 2 possible solutions
1) add rm (like we do with make) introducing potentiol new issues with virus scanners removing it (like make)
2) delete the target folder like we do in the workaround

jonnieZG, 2016-06-09:
You can also install the GNU CoreUtils for Windows from http://gnuwin32.sourceforge.net/packages/coreutils.htm and add path to .../bin directory to the Win Environment System Path.

jantje, 2016-06-09:
@jonnieZG thanks for the input. We are trying to get the product in a "run out of the box" state. So having to install other things and/or changing the path is a second option.
But I'll look into the option to download it as a tool.

closed 2016-08-01 by jantje