Sloeber/arduino-eclipse-plugin #466: xtensa-lx106-elf-g++: not found

myrtleTree33, 2016-05-31:
I am trying to build a basic .ino Arduino file on the ESP8266 (NodeMCU v1.0) using the V3 version.

However, it seems to compile to C++ reporting the following:

/bin/sh: 1: /opt/eclipse/arduinoPlugin/tools/esp8266/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++: not found

Seems that the g++ plugin is missing?  Help?

jantje, 2016-05-31:
I decided to no longer respond to github issues -and other questions- for people not supporting the project.
As a contributor your issues are treated. As a patron you can post a message on the patreon forum stating you support a github issue. This allows me to support people who support me (starting from as low as 1$ a month exclusive VAT).
https://www.patreon.com/posts/3700469

jantje, 2016-08-20:
closing as duplicate

closed 2016-08-20 by jantje