Sloeber/arduino-eclipse-plugin #467: Compiling error: Release/syscalls.c.o: No such file or directory

ShenggaoZhu, 2016-06-01:
When compiling the project for the first time (on Mac), an error occurs:

> arm-none-eabi-g++: error: .../Release/syscalls.c.o: No such file or directory

To avoid this error, I have to copy the file `Release/core/syscalls.c.o` to `Release/syscalls.c.o`.

Is this problem easy to fix via some configuration? Thanks!

jantje, 2016-06-01:
I decided to no longer respond to github issues -and other questions- for people not supporting the project.
As a contributor your issues are treated. As a patron you can post a message on the patreon forum stating you support a github issue. This allows me to support people who support me (starting from as low as 1$ a month exclusive VAT).
https://www.patreon.com/posts/3700469

closed 2016-08-20 by jantje

kumarsatish2012, 2019-11-08:
how to resolve this error????

arm-none-eabi-g++: error: C:\Users\SATISH\AppData\Local\Temp\arduino_build_107676/core/syscalls.c.o: No such file or directory