Sloeber/arduino-eclipse-plugin #468: Oscilloscope not working or is it just me

universpharmacie, 2016-06-01:
Tried to get that oscilloscope working - without success. Tried some of the examples which I could find on google - none worked. When digging for more info I discovered that the simplot website is not longer online. Could someone point me towards a working example code please?

EdwinCroissant, 2016-06-01:
I used it a couple of days ago. The library can be found here: https://github.com/jantje/ArduinoLibraries/tree/master/simplot

universpharmacie, 2016-06-01:
Thank you very much. I wasn't aware of that library, I tried some simple examples using Serial.write() found on the Arduino support forums. But since I don't use Arduinos but Teensys, I obviously had to reverse the byte order. I'll give it a new try tonight when I'll be at home.

jantje, 2016-06-01:
note that you can move the scale and adopt the scale with your mouse. Just drag and drop.
And the library works with Teensy 3.1.

jantje, 2016-06-10:
@universpharmacie Did it work by now?

universpharmacie, 2016-06-10:
Totally under water at my day job, actually doing 16h/day, no time to try out before the upcoming weekend. You may close here if you prefer, since my problem (low-high byte swap) was clearly identified.

jantje, 2016-06-10:
I had a workshop and there were plenty of mac users. The scope works fine with the scope library. the only problem on mac is that you can not zoom (left drag).
I'll close the issue. If things turn out bad you can reopen it.

closed 2016-06-10 by jantje