Sloeber/arduino-eclipse-plugin #472: installing the plugin killed my eclipse

piersh, 2016-06-05:
fresh install of mars.2, fresh install of plugin v3.

"we have found programs in the path that might conflict with our external builder.
The conflicting programs are bash and sh.
The program might still function but if you get strange build errors you know where to look"

eclipse just quits after this.

what now?

jantje, 2016-06-05:
Thanks for fixing this yourself :-)

Wazalogic, 2016-08-24:
I have the same problem. What was the fix?

rlogiacco, 2016-08-24:
@Wazalogic I believe @jantje was referring to the fact you closed the issue...
In any case, as the message says, both `bash` and `sh` are creating problems to the build, though they have not been reported as making Eclipse crash.
First advice I can give you is to identify the software which added `bash` or `sh` to your system path and see if you can temporarily or permanently eliminate that path inclusion: if you can, just do it and try again.

wimjongman, 2016-10-31:
This is still a problem and is very common. Everyone that has GIT installed will run into this issue. 

With some help of the net [1] I have "created" a batch script that I use to start **Sloeber**. Copy this script, call it sloeber.cmd and put it next to the sloeber-ide.exe:

`
@echo off
setlocal EnableDelayedExpansion
set path
set $line=%path%
set $line=%$line: =#%
set $line=%$line:;= %
set $line=%$line:)=^^)%

REM Here we filter all dirs with "Git" in the name. 
for %%a in (%$line%) do echo %%a | find /i "Git" || set $newpath=!$newpath!;%%a
set $newpath=!$newpath:#= !
set $newpath=!$newpath:^^=!
set path=!$newpath:~1!

REM Run Sloeber, Run!
sloeber-ide.exe
` 

[1] http://stackoverflow.com/questions/21289762/remove-unwanted-path-name-from-path-variable-via-batch#21292905

jantje, 2016-10-31:
@wimjongman 
If this is a good solution shouldn't we add this to the product download?

benilevIntel, 2016-10-31:
Same here.
I have mingw installed

beniloew, 2016-10-31:
Same here.
I have bot GIT and MinGW installed

jantje, 2016-10-31:
This code removes both git and mingw from the path before starting Sloeber
```
@echo off
setlocal EnableDelayedExpansion
set path
set $line=%path%
set $line=%$line: =#%
set $line=%$line:;= %
REM here we replace mingw with git
set $line=%$line:Mingw=git%
set $line=%$line:)=^^)%

REM Here we filter all dirs with "Git" in the name. 
for %%a in (%$line%) do echo %%a | find /i "git" || set $newpath=!$newpath!;%%a
set $newpath=!$newpath:#= !
set $newpath=!$newpath:^^=!
set path=!$newpath:~1!

REM Run Sloeber, Run!
sloeber-ide.exe
```

beniloew, 2016-11-01:
This indeed solves the issue.
Thanks a lot!

wimjongman, 2016-11-02:
Would tinkering with system property java.library.path work? 

-Djava.library.path=some/path

jantje, 2016-11-03:
>-Djava.library.path=some/path  

As far as I can see this is the java library path and it is not related to the system path

tomkcook, 2016-12-01:
This issue has got a lot worse since Microsoft released Windows Subsystem for Linux, which puts a `bash.exe` in C:\Windows\System32.  It's not clear what can be done; it definitely breaks builds (it's a 'proper' Linux bash, so doesn't understand backslashes in paths - I get build errors such as `.librariesSPIsrcSPI.cpp.o: No such file or directory` which has clearly had the backslashes removed) and removing system32 from the system path is not really a viable option.

jantje, 2016-12-01:
I have no idea right now. I need to look closer into the issue and it is pretty far away from my comfort zone ;-).
What I do know:
Sloeber does not call bash or sh directly. As the problem happens during build I guess it is make that does it.
Maybe another make would fix the issue or adding a good bash/sh in the path.

Note: I don't add make to the path. The default make command is ${JANTJE.MAKE_LOCATION}make on non-windows os'es ${JANTJE.MAKE_LOCATION} resolves to nothing on windows it resolves to the correct location.

jantje, 2016-12-01:
Replacing the make of the one delivered with mingw32 seems to fix the issue :-)

tomkcook, 2016-12-02:
I've also found that starting eclipse from an environment that doesn't include system32 or mingw also works.

jantje, 2016-12-02:
V4 no longer downloads make.exe from a ftp site but from a zip file from my site. This means I have more control over the content after releasing.
V4 still contains the warning message but it should no longer experience the problem.
We should start testing V4 :-)

jantje, 2016-12-06:
Seems the mingw make has problems with the defines with spaces. Those boards won't work untill I fixed the make file generator.

closed 2017-01-17 by jantje