Sloeber/arduino-eclipse-plugin #476: Compiler Issue - wrong location for include file

epgibbons, 2016-06-07:
Compilation problem:

```

Starting upload
using arduino loader
Starting reset using DTR toggle process
Toggling DTR
Continuing to use COM4
Ending reset

Launching D:\Users\edwin\Documents\Arduino\V3.0_win64.2016-03-31_12-43-53\eclipseArduino\arduinoPlugin\tools\arduino\avrdude\6.0.1-arduino5/bin/avrdude -CD:\Users\edwin\Documents\Arduino\V3.0_win64.2016-03-31_12-43-53\eclipseArduino\arduinoPlugin\tools\arduino\avrdude\6.0.1-arduino5/etc/avrdude.conf -patmega328p -carduino -PCOM4 -b57600 -D -Uflash:w:D:\Users\edwin\Documents\Arduino\workspace\Initial/Release/Initial.hex:i 
Output:

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e950f
avrdude: reading input file "D:\Users\edwin\Documents\Arduino\workspace\Initial/Release/Initial.hex"
avrdude: can't open input file D:\Users\edwin\Documents\Arduino\workspace\Initial/Release/Initial.hex: No such file or directory
avrdude: read from file 'D:\Users\edwin\Documents\Arduino\workspace\Initial/Release/Initial.hex' failed

```
How to reproduce:

Just installed the Arduino IDE product and created my first cpp sketch to read a potentiometer.

```

// Do not remove the include below
#include "Initial.h"

const int POT=0;
//The setup function is called once at startup of the sketch
void setup()
{
	Serial.begin(9600);
}

// The loop function is called in an endless loop
void loop()
{
	int val = analogRead(POT);
	Serial.println(val);
	delay(500);
}

```

The Initial.h file has path /Initial/Initial.h not \Initial/Release as expected by the compiler.

wimjongman, 2016-06-07:
Nice to see you here Paul!

closed 2016-06-07 by wimjongman

jantje, 2016-06-07:
ignore ;-)