Sloeber/arduino-eclipse-plugin #478: #define statements are not recognized by included libraries (e.g. mysensors)

DerTomm, 2016-06-15:
Hi Jantje,

first of all let me thank you and all the other contributors for the great third release of EclipseArduino. I really like it and appreciate all of your work!

I just discovered one (I guess minor) issue today when I tried to compile the current MySensors SerialGateway example (https://github.com/mysensors/Arduino/tree/development/libraries/MySensors/examples/GatewaySerial) with Eclipse.

Within Arduino IDE 1.6.9 this example compiles without any problems. But when I import the same example in Eclipse the #define statements are not recognized by the main library MySensors.h.
This block does not see "MY_GATEWAY_SERIAL" defined:

`#if defined(MY_GATEWAY_SERIAL) || defined(MY_GATEWAY_W5100) || defined(MY_GATEWAY_ENC28J60) || defined(MY_GATEWAY_ESP8266) || defined(MY_GATEWAY_MQTT_CLIENT)
	#define MY_GATEWAY_FEATURE
	#define MY_IS_GATEWAY (true)`

But in the ino file there is the statement:
`#define MY_GATEWAY_SERIAL`

Do you have any ideas why the define statement is not processed properly here? The interesting thing is that Eclipse shows me the defined value in the mouse over popup - but the compiler seems not to recognize it...

jantje, 2016-06-15:
What you are describing is a known limitation.
Read this article http://blog.baeyens.it/#post13 and look at the generated .ino.cpp file in your project to get a better understanding of wat is going on.
use the project properties->arduino->compile options to add -DMY_GATEWAY_SERIAL to work around this.

jantje, 2016-11-09:
I have written another blog on how to handle this situation https://www.patreon.com/posts/defines-before-6795548

dgolda, 2017-01-04:
I suppose I have the same problem while using Souliss "insketch" configuration: https://github.com/souliss/souliss/wiki/inSketch

For example in https://github.com/souliss/souliss/blob/friariello/examples/usart/e01_GarageDoor_n1/e01_GarageDoor_n1.ino there are defines in lines 39-42 which are not visible in #include "Souliss.h" below.

jantje, 2017-01-04:
This should be fixed in the latest V4.
It would be great if you could test and report results.

dgolda, 2017-01-07:
I can confirm, in 4.0.0-20170106.184340-70-win32.win32.x86_64 works for me.
Thank you, jantje!

closed 2017-01-17 by jantje