Sloeber/arduino-eclipse-plugin #479: Can't upload to Yun from V3 plugin, but upload from Arduino IDE 1.6.10 working.

emepetres, 2016-06-17:
Hi, I am using Linux x64 (archlinux) and Eclipse Neon, connecting my Arduino Yun in an USB3 slot. 

While upload using Arduino IDE nightly (6-Jun) is working, V3 plugin seem to not be able to connect to the Yun through the /dev/ttyACM0 port and then upload the file, giving avrdude the error _

[ArduinoIDE.txt](https://github.com/jantje/arduino-eclipse-plugin/files/320274/ArduinoIDE.txt)
[V3Plugin.txt](https://github.com/jantje/arduino-eclipse-plugin/files/320273/V3Plugin.txt)

"Connecting to programmer: .avrdude: butterfly_recv(): programmer is not responding"_.

Everything is up to date at the time of writing (17-06-2016)

I attach the logs from Arduino IDE and V3 plugin for reference.

jantje, 2016-06-17:
From the logs you send it is clear that the yun did not reset which makes the upload fail.
I never got the leonardo to work on linux64 bit. But as it is a yun why don't you use upload over tcpip?

emepetres, 2016-06-17:
Upload over tcp/ip (wifi) worked like a charm. Anyway it would be nice to be able to use the serial port as well.

Thank you very much, great plugin!

jantje, 2016-06-17:
>  Anyway it would be nice to be able to use the serial port as well.

I know. But as I seem to be unable to get it working on my linux machine (I mean not in the Arduino ide nor in Sloeber) I can not debug it.
From the logs you have send I see there is a reset problem.

jantje, 2017-10-08:
There have been lots of changes in this area lately.
Do you still have the issue?

closed 2018-03-04 by jantje