Sloeber/arduino-eclipse-plugin #480: Incorrect environment variable for DUE makes it so sketches cannot be uploaded

mikeshub, 2016-06-20:
The sketch uploads to a mega 2560 through arduino eclipse and to the DUE through 1.6.9. When the due is selected the sketch cannot be uploaded because this is the command used by arduino eclipse:

C:\eclipseArduino\arduinoPlugin\tools\arduino\bossac\1.6.1-arduino/bossac.exe --port= -U false -e -w -b C:\Users\Hammer\Documents\eclipseArduino\TestSketch/Debug/TestSketch.bin -R 

From arduino 1.6.9 it uses:

C:\Users\Hammer\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.6.1-arduino/bossac.exe -i -d --port=COM8 -U false -e -w -v -b C:\Users\Hammer\AppData\Local\Temp\builddc6fc8b25f29d94c93a61449a3e1a75d.tmp/Pololu_Open_IMU.ino.bin -R 

Clearly the main problem is the lack of com port from arduino eclipse. Looking at the environment variables 

I have 

A.TOOLS.BOSSAC.UPLOAD.PATTERN

as 

"${A.TOOLS.BOSSAC.PATH}/${A.TOOLS.BOSSAC.CMD}" ${A.TOOLS.BOSSAC.UPLOAD.VERBOSE} --port=${A.TOOLS.BOSSAC.SERIAL.PORT.FILE} -U ${A.UPLOAD.NATIVE_USB} -e -w ${A.TOOLS.BOSSAC.UPLOAD.VERIFY} -b "${A.BUILD.PATH}/${A.BUILD.PROJECT_NAME}.bin" -R

A.TOOLS.BOSSAC.SERIAL.PORT.FILE is not an environment variable. I can neither change the A.TOOLS.BOSSAC.UPLOAD.PATTERN variable to use the correct port and I cannot create an environment variable called A.TOOLS.BOSSAC.SERIAL.PORT.FILE permanently. When I either edit the A.TOOLS.BOSSAC.UPLOAD.PATTERN variable or create A.TOOLS.BOSSAC.SERIAL.PORT.FILE they do not remain set. I hit apply OK and then when I open them again the changes have been discarded.

jantje, 2016-06-20:
Thanks for the good research. I'll  have a look.

jantje, 2016-06-21:
seems like 1.6.8 has this problem as well.

jantje, 2016-06-21:
I'm confused her.
The latest version for due is 1.6.8 and is using 
`tools.bossac.upload.pattern="{path}/{cmd}" {upload.verbose} --port={serial.port.file} -U {upload.native_usb} -e -w {upload.verify} -b "{build.path}/{build.project_name}.bin" -R
`
Where did you get the version with `TOOLS.BOSSAC.SERIAL.PORT.FILE` ?

closed 2016-08-01 by jantje