Sloeber/arduino-eclipse-plugin #481: Doesn't find Java 1.8 OS X

trevorsmedley, 2016-06-21:
Hi,

Just trying to get this up and running, and I get the error:  Version 1.6.0_65 of the JVM is not suitable for this product.

I have Java Version 8 Update 73 installed (this is what Safari shows, for example, when I test my java version using https://www.java.com/verify/). I also have the "legacy" JVM 1.6 installed, which is what seems to be causing the problem.

How can I get eclipseArduino to see the new Java version rather than the old?

Thanks, Trevor

trevorsmedley, 2016-06-21:
Never mind. It's an issues with Eclipse looking for Java Developer Kit version, rather than the runtime. Resolved.

closed 2016-06-21 by trevorsmedley

jantje, 2016-06-21:
thanks for the good work :-)