Sloeber/arduino-eclipse-plugin #482: Atmega328PB doesn't compile: !MESSAGE failed to find core reference: arduino:arduino

mquade, 2016-06-23:
I'm using Nightly and added an additional board file from https://github.com/watterott/ATmega328PB-Testing --> https://github.com/watterott/ATmega328PB-Testing/raw/master/package_m328pb_index.json
I can select the board in the board manager, but compiling doesn't work. So I've replaced the toolchain as described on his site 
> The current Arduino avr-gcc toolchain (4.8.1) does not support the ATmega328PB. To get support you can replace it with the below avr-gcc toolchain (4.9.2 from Atmel or Arduino) and update the files in /arduino/hardware with the files from this repository under hardware.

Just putting the files into my eclipse folder under ...\eclipse-cpp-mars-2-win32-x86_64\eclipse\arduinoPlugin\tools\arduino\avr-gcc didn't work, the name of the directory seems to be hardcoded, so I had to rename the new avr-gcc-directory back to 4.8.1-arduino5. 
With this new toolchain, the standardboards (Arduino Uno etc.) compile/work. 
But if I select the ATmega328PB I get following Error in the Log:
```
!ENTRY it.baeyens.arduino.core 4 0 2016-06-23 08:26:39.970
!MESSAGE failed to find core reference: arduino:arduino

```
Compilerwindow:
```

08:26:35 **** Incremental Build of configuration Release for project test111 ****
"C:\\Users\\xxx\\Downloads\\eclipse-cpp-mars-2-win32-x86_64\\eclipse\\arduinoPlugin\\tools\\make\\make" all 
'Building file: ../test111.cpp'
'Starting C++ compile'
I"C:\Users\xxx\Downloads\eclipse-cpp-mars-2-win32-x86_64\eclipse\arduinoPlugin\packages\m328pb\hardware\atmega328pb\avr\cores\arduino" -I"C:\Users\xxx\Downloads\eclipse-cpp-mars-2-win32-x86_64\eclipse\arduinoPlugin\packages\m328pb\hardware\atmega328pb\avr\variants\atmega328pb" -MMD -MP -MF"test111.cpp.d" -MT"test111.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "../test111.cpp"  "test111.cpp.o"   -Wall
subdir.mk:18: recipe for target 'test111.cpp.o' failed
process_begin: CreateProcess(NULL, "IC:\Users\xxx\Downloads\eclipse-cpp-mars-2-win32-x86_64\eclipse\arduinoPlugin\packages\m328pb\hardware\atmega328pb\avr\cores\arduino" "-IC:\Users\xxx\Downloads\eclipse-cpp-mars-2-win32-x86_64\eclipse\arduinoPlugin\packages\m328pb\hardware\atmega328pb\avr\variants\atmega328pb" -MMD -MP -MFtest111.cpp.d -MTtest111.cpp.o -D__IN_ECLIPSE__=1 -x c++ ../test111.cpp test111.cpp.o -Wall, ...) failed.
make (e=2): Das System kann die angegebene Datei nicht finden.

make: [test111.cpp.o] Error 2 (ignored)
'Finished building: ../test111.cpp'
' '
'Starting archiver'
arduino.ar 
makefile:75: recipe for target 'arduino.ar' failed
process_begin: CreateProcess(NULL, arduino.ar, ...) failed.
make (e=2): Das System kann die angegebene Datei nicht finden.

make: *** [arduino.ar] Error 2

08:26:35 Build Finished (took 180ms)
```

mquade, 2016-06-23:
I've edited the boards.txt and removed the atmega328pbcc.build.core=arduino**:arduino**.
Now I'm running in other problems... 

`12:45:48 **** Incremental Build of configuration Release for project test1 ****
"C:\\Users\\xxx\\Downloads\\eclipse-cpp-mars-2-win32-x86_64 (1)\\eclipse\\arduinoPlugin\\tools\\make\\make" all 
'Building file: ../test1.cpp'
'Starting C++ compile'
"C:\Users\xxx\Downloads\eclipse-cpp-mars-2-win32-x86_64 (1)\eclipse\arduinoPlugin\tools\arduino\avr-gcc\4.8.1-arduino5/bin/avr-g++" -c -g -Os -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega328pb -DF_CPU=8000000L -DARDUINO=10609 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR     -I"C:\Users\xxx\Downloads\eclipse-cpp-mars-2-win32-x86_64 (1)\eclipse\arduinoPlugin\packages\arduino\hardware\avr\1.6.11\cores\arduino" -I"C:\Users\xxx\Downloads\eclipse-cpp-mars-2-win32-x86_64 (1)\eclipse\arduinoPlugin\packages\arduino\hardware\avr\1.6.11\variants\atmega328pb" -MMD -MP -MF"test1.cpp.d" -MT"test1.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "../test1.cpp"  -o  "test1.cpp.o"   -Wall  -Wall
In file included from c:\users\xxx\downloads\eclipse-cpp-mars-2-win32-x86_64 (1)\eclipse\arduinoplugin\tools\arduino\avr-gcc\4.8.1-arduino5\avr\include\avr\pgmspace.h:90:0,
                 from C:\Users\xxx\Downloads\eclipse-cpp-mars-2-win32-x86_64 (1)\eclipse\arduinoPlugin\packages\arduino\hardware\avr\1.6.11\cores\arduino/Arduino.h:28,
                 from ../test1.h:9,
                 from ../test1.cpp:2:
c:\users\xxx\downloads\eclipse-cpp-mars-2-win32-x86_64 (1)\eclipse\arduinoplugin\tools\arduino\avr-gcc\4.8.1-arduino5\avr\include\avr\io.h:623:6: warning: #warning "device type not defined" [-Wcpp]
 #    warning "device type not defined"
      ^
In file included from C:\Users\xxx\Downloads\eclipse-cpp-mars-2-win32-x86_64 (1)\eclipse\arduinoPlugin\packages\arduino\hardware\avr\1.6.11\cores\arduino/USBAPI.h:25:0,
                 from C:\Users\xxx\Downloads\eclipse-cpp-mars-2-win32-x86_64 (1)\eclipse\arduinoPlugin\packages\arduino\hardware\avr\1.6.11\cores\arduino/Arduino.h:225,
                 from ../test1.h:9,
                 from ../test1.cpp:2:
c:\users\xxx\downloads\eclipse-cpp-mars-2-win32-x86_64 (1)\eclipse\arduinoplugin\tools\arduino\avr-gcc\4.8.1-arduino5\avr\include\avr\eeprom.h:41:3: warning: #warning "Device does not have EEPROM available." [-Wcpp]
 # warning "Device does not have EEPROM available."
   ^
'Finished building: ../test1.cpp'
' '
'Building file: C:/Users/xxx/Downloads/eclipse-cpp-mars-2-win32-x86_64 (1)/eclipse/arduinoPlugin/packages/arduino/hardware/avr/1.6.11/cores/arduino/CDC.cpp'
'Starting C++ compile'
"C:\Users\xxx\Downloads\eclipse-cpp-mars-2-win32-x86_64 (1)\eclipse\arduinoPlugin\tools\arduino\avr-gcc\4.8.1-arduino5/bin/avr-g++" -c -g -Os -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega328pb -DF_CPU=8000000L -DARDUINO=10609 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR     -I"C:\Users\xxx\Downloads\eclipse-cpp-mars-2-win32-x86_64 (1)\eclipse\arduinoPlugin\packages\arduino\hardware\avr\1.6.11\cores\arduino" -I"C:\Users\xxx\Downloads\eclipse-cpp-mars-2-win32-x86_64 (1)\eclipse\arduinoPlugin\packages\arduino\hardware\avr\1.6.11\variants\atmega328pb" -MMD -MP -MF"core/CDC.cpp.d" -MT"core/CDC.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "C:/Users/xxx/Downloads/eclipse-cpp-mars-2-win32-x86_64 (1)/eclipse/arduinoPlugin/packages/arduino/hardware/avr/1.6.11/cores/arduino/CDC.cpp"  -o  "core/CDC.cpp.o"   -Wall  -Wall
In file included from c:\users\xxx\downloads\eclipse-cpp-mars-2-win32-x86_64 (1)\eclipse\arduinoplugin\tools\arduino\avr-gcc\4.8.1-arduino5\avr\include\avr\pgmspace.h:90:0,
                 from C:/Users/xxx/Downloads/eclipse-cpp-mars-2-win32-x86_64 (1)/eclipse/arduinoPlugin/packages/arduino/hardware/avr/1.6.11/cores/arduino/USBAPI.h:24,
                 from C:/Users/xxx/Downloads/eclipse-cpp-mars-2-win32-x86_64 (1)/eclipse/arduinoPlugin/packages/arduino/hardware/avr/1.6.11/cores/arduino/CDC.cpp:19:
c:\users\xxx\downloads\eclipse-cpp-mars-2-win32-x86_64 (1)\eclipse\arduinoplugin\tools\arduino\avr-gcc\4.8.1-arduino5\avr\include\avr\io.h:623:6: warning: #warning "device type not defined" [-Wcpp]
 #    warning "device type not defined"
      ^
In file included from C:/Users/xxx/Downloads/eclipse-cpp-mars-2-win32-x86_64 (1)/eclipse/arduinoPlugin/packages/arduino/hardware/avr/1.6.11/cores/arduino/USBAPI.h:25:0,
                 from C:/Users/xxx/Downloads/eclipse-cpp-mars-2-win32-x86_64 (1)/eclipse/arduinoPlugin/packages/arduino/hardware/avr/1.6.11/cores/arduino/CDC.cpp:19:
c:\users\xxx\downloads\eclipse-cpp-mars-2-win32-x86_64 (1)\eclipse\arduinoplugin\tools\arduino\avr-gcc\4.8.1-arduino5\avr\include\avr\eeprom.h:41:3: warning: #warning "Device does not have EEPROM available." [-Wcpp]
 # warning "Device does not have EEPROM available."
   ^
In file included from c:\users\xxx\downloads\eclipse-cpp-mars-2-win32-x86_64 (1)\eclipse\arduinoplugin\tools\arduino\avr-gcc\4.8.1-arduino5\avr\include\avr\io.h:99:0,
                 from c:\users\xxx\downloads\eclipse-cpp-mars-2-win32-x86_64 (1)\eclipse\arduinoplugin\tools\arduino\avr-gcc\4.8.1-arduino5\avr\include\avr\pgmspace.h:90,
                 from C:/Users/xxx/Downloads/eclipse-cpp-mars-2-win32-x86_64 (1)/eclipse/arduinoPlugin/packages/arduino/hardware/avr/1.6.11/cores/arduino/USBAPI.h:24,
[... cut ...]
c:\users\xxx\downloads\eclipse-cpp-mars-2-win32-x86_64 (1)\eclipse\arduinoplugin\tools\arduino\avr-gcc\4.8.1-arduino5\avr\include\avr\wdt.h:526:55: error: 'WDE' was not declared in this scope
     [WDCE_WDE]  "n"  ((uint8_t)(_BV(_WD_CHANGE_BIT) | _BV(WDE)))
                                                       ^
make: *** [core/CDC.cpp.o] Error 1
core/core/subdir.mk:98: recipe for target 'core/CDC.cpp.o' failed

12:45:49 Build Finished (took 433ms)

`

I wll try further but I don't really know where to start here...

jantje, 2016-06-23:
This is clearly a duplicate of #421 
There is a workaround described in #421 
I assume the actions you did with the toolchain changes may have made things worse.

mquade, 2016-06-23:
Yeah, It works!
I did the steps of #421 , and additional I had to copy the files of the provided hardware.zip file to the right locations:
\Downloads\hardware\tools... ---> Everything goes to the correspondig folders in \eclipse\arduinoPlugin\tools\arduino\avr-gcc\4.8.1-arduino5\avr\lib\avr5

May bee this will help some one :)

jantje, 2016-06-23:
Cool. I'll have a look to see whether there is an easy solution for #421

closed 2016-06-23 by jantje