Sloeber/arduino-eclipse-plugin #483: Serial minitor missing baud rates

jonbarril, 2016-06-23:
Trying this eclipse plugin for the first time.  Using V3.  Serial monitor baud rates 230400 and 250000, which are available in the Arduino IDE, are missing.  I'd have to change all of my ino files to work with the plugin.  Is there some reason for this?  Is there an easy workaround?

jantje, 2016-06-23:
The easy work around is to just click the open serial monitor icon when the project you want to connect to is selected. The baud rate is selected based on the `Serial.begin([Your baud rate here])`; located in `void setup()` and the com port is the one you use to upload.
These baud rates are added by the arduino core team and no one requested them. I'll add them.

jantje, 2016-06-23:
It should be available in the nightly tomorrow.
Can you report whether it works?

jonbarril, 2016-06-24:
Not sure if I follow what you are saying.  I tried using the monitor on 
a board running an app I built using the Arduino IDE. Since I could not 
select 230400 as the speed the monitor was printing gibberish.

In any case, I updated to the nightly build and now see all the speeds 
listed.  I will try it on an actual board tomorrow.

Thanks for all your work doing the plugin.
--jon

jantje, 2016-06-24:
The work around is actually the easiest way to open a serial connection. No need to select com port nor baud rate :-)
Watch this video to understand how it works. https://www.youtube.com/watch?v=x_JKcvqpxq8&list=PLCSjf7QmP3TDPSjT_5UIp0DHXzKdhIQsF&index=2

jonbarril, 2016-06-26:
Got busy with other things.  Will watch the video when I get back to 
arduino in eclipse.  I tried the monitor with my existing app and, with 
the 230400 selection, it works.

BTW: I have a number of existing eclipse/arduino projects where I use 
eclipse and CDT with no special arduino stuff.  How to I c onvert/import 
those for use with your plugin? (If the answer is "look at the video" 
then shame on me;-)

--jon

jonbarril, 2016-06-26:
BTW: I tried to create a test project via the plugin.  I see that it 
created a bunch of stuff in the project.  In the Arduino config I added 
the path to by private libraries (root dir with library dirs underneath, 
like Arduino IDE wants them).  When I tried the verify the project build 
I got a dialog containing...

Failed to build the code
Errors occurred during the build.
Errors running builder 'Scanner Configuration Builder' on project 
'MySketch'.
java.lang.NullPointerException

--jon

> > The work around is actually the easiest way to open a serial > 
connection. No need to select com port nor baud rate :-) Watch this > 
video to understand how it works. > 
https://www.youtube.com/watch?v=x_JKcvqpxq8&list=PLCSjf7QmP3TDPSjT_5UIp0DHXzKdhIQsF&index=2 
 > > >
—
<https://github.com/jantje/arduino-eclipse-plugin/issues/483#issuecomment-228320841>, 
 > or mute the thread > 
<https://github.com/
 >

jantje, 2016-06-26:
Thanks fro confirming it works.
As to the other problems: watch the video's

closed 2016-08-01 by jantje