Sloeber/arduino-eclipse-plugin #484: Neon Support

wimjongman, 2016-06-24:
Provide neon support for Sloeber.

jantje, 2016-06-24:
That is quick :-)

wimjongman, 2016-06-24:
All good things arrive quickly.

compihu, 2016-06-26:
Plugin cannot be installed to Neon from the nightly update site at the moment:

Cannot complete the install because one or more required items could not be found.
  Software being installed: Arduino Eclipse Extensions 3.0.0.201606260209 (it.baeyens.arduino.feature.feature.group 3.0.0.201606260209)
  Missing requirement: Common 3.0.0.201606260209 (it.baeyens.arduino.common 3.0.0.201606260209) requires 'bundle org.eclipse.wb.core.lib 0.0.0' but it could not be found
  Cannot satisfy dependency:
    From: Arduino Eclipse Extensions 3.0.0.201606260209 (it.baeyens.arduino.feature.feature.group 3.0.0.201606260209)
    To: it.baeyens.arduino.common [3.0.0.201606260209]

jantje, 2016-06-26:
tested this here on windows 64 bit. 
I have the same problem.

jantje, 2016-06-26:
The product states it is eclipse 4.5 which is Mars and not neon.

compihu, 2016-06-27:
Thanks Jantje, Is there a way to keep the nightly installable in Neon release? (current ubuntu LTS is incompatible with Mars)

wimjongman, 2016-06-27:
Good work requires time.. ;)

> It can be that the build script uses neon explicitly

Windowbuilder is no longer part of the release train. I will remove it from the requirements if possible.

jantje, 2016-06-27:
@compihu 
as you see Wim is far better at this than I am :-)
Thanks Wim

wimjongman, 2016-06-27:
You're welcome. Neon should now be the default. Please test the product and the update site once available.

jantje, 2016-06-27:
I will :-)

jantje, 2016-06-28:
You can now install the plugin in neon :-)
The product is still mars but I think this is because jenkins asked to build mars. Changed that so well see next build.

jantje, 2016-06-28:
I ran a special build. The nightly is now neon to :-)

closed 2016-08-01 by jantje