Sloeber/arduino-eclipse-plugin #487: Connecting to programmer: .avrdude: butterfly_recv(): programmer is not responding" error.

davidlevinson, 2016-06-24:
I have a clean installation of Eclipse Neon + Eclipse Arduino v3 Plugin + Leonardo board + MacOS 10 + Arduino IDE 1.6.5.

There is an issue in the "reset" process during the upload. Whatever steps are happening in "Starting reset using 1200bps touch process", this is not correctly resetting the board. I get a "Connecting to programmer: .avrdude: butterfly_recv(): programmer is not responding" error.

In order to get a sketch to upload I need to click the reset button on the Leonardo shorty after seeing the first PORTS output. 

Perhaps there is a way to skip this step or improve this logic?

jantje, 2016-06-24:
I decided to no longer respond to github issues -and other questions- for people not supporting the project.
As a contributor your issues are treated. As a patron you can post a message on the patreon forum stating you support a github issue. This allows me to support people who support me (starting from as low as 1$ a month exclusive VAT).
https://www.patreon.com/posts/3700469

jirisemecky, 2016-08-18:
David, were you able to resolve the problem?

jantje, 2016-09-03:
Closing due to no sponsor

closed 2016-09-03 by jantje